genodelabs/genode #2354: libc: cannot override stack size

cproc, 2017-03-27:
The test program in commit https://github.com/cproc/genode/commit/c5b54e1716107c77ba256250b56fe576b9e9ca2a overrides the weak 'Libc::Component::stack_size()' function, but it doesn't get called. The 'nm' tool doesn't even show the symbol in the binary.

chelmuth, 2017-03-28:
The following patch fixed the issue for me.
```diff
diff --git a/repos/base/lib/symbols/ld b/repos/base/lib/symbols/ld
index 11b6568..0814126 100644
--- a/repos/base/lib/symbols/ld
+++ b/repos/base/lib/symbols/ld
@@ -38,6 +38,8 @@
 # under the terms of the GNU Affero General Public License version 3.
 #
 
+_ZN4Libc9Component10stack_sizeEv W
+
 _Unwind_Complete T
 _Unwind_DeleteException T
 _Unwind_Resume T
```
The reason is that *libc.abi.so* does not declare its implementation of `Libc::Component::stack_size()` as weak and, so, it's not looked up in the binary. Finally, the link of the target garbage collects the unused function in *test-libc_stack_size*.

chelmuth, 2017-03-28:
Thanks for the correct patch ;-)

commit 2a64736e (2017-03-28): libc: add 'Libc::Component::stack_size()' to the ABI symbol file

Fixes #2354
commit 4799d783 (2017-04-11): libc: add 'Libc::Component::stack_size()' to the ABI symbol file

Fixes #2354
commit a62f09e8 (2017-05-02): libc: add 'Libc::Component::stack_size()' to the ABI symbol file

Fixes #2354
closed 2017-05-31 by chelmuth