genodelabs/genode #2360: cli_monitor not working on linux

BorisMulder-CSL, 2017-03-31:
I was playing around with cli_monitor and discovered it does not run on linux at all. On nova+qemu it does run normally, but when I change the kernel to linux it gives me the errors:

Kernel: [1063] lx_sendmsg failed with -9 in lx_reply() reply_socket=1001
Kernel: [1063] lx_sendmsg failed with -9 in lx_reply() reply_socket=-24
Kernel: [1063] lx_sendmsg failed with -88 in lx_reply() reply_socket=0

After this, the whole system hangs.
I tried running it with and without subsystems, but it gives the same errors. The cli_monitor's config looks as follows (without subsystems):

<start name="cli_monitor" priority="-3">
	<resource name="RAM" quantum="30M"/>
	<config verbose="yes">
		<vfs/>
	</config>
	<route>
		<any-service>
			<child name="cli_terminal"/>
			<child name="wm"/>
			<parent/> <any-child/>
		</any-service>
	</route>
</start>

chelmuth, 2017-03-31:
Please consider that each Genode capability is implemented by a open file in Linux. Therefore, you have to increase the open file limit significantly. Please investigate with `ulimit -Hn` and `ulimit -Sn`,

BorisMulder-CSL, 2017-03-31:
Thanks, that worked.
May I ask why an empty cli_monitor needs that many caps? Or did linux just happen to run out at this point?

closed 2017-03-31 by BorisMulder-CSL

chelmuth, 2017-03-31:
You're right that cli_monitor itself does not need that many capabilities (speak per-process file descriptors). The problem is that core which manages all resources needs open files for those resources. In this case I'm talking about file descriptors for RAM dataspaces for data segments, heap and stacks as well as binaries and shared libs.

You may investigate what happens by running `watch -n 0.5 'ls -1 /proc/$(pidof ./core)/fd | wc -l'` in a terminal while playing with *run/demo*.