genodelabs/genode #2361: noux bash: unable to execute programs

BorisMulder-CSL, 2017-03-31:
When I try to run bash in noux and execute a program, I get the following error on nova:

[init -> noux -> /bin/bash] pthread_main_np_int: pthread_main_np called, not implemented
Then bash gives an error: A NULL argv[0] was passed through an exec system call.

When I do the same on Linux, I get:

Kernel: ipc_cap_data nullptr
In this case bash stops working at all.

This can be verified by running noux_bash.run.
Besides, on linux you might have to remove the RTC driver from the noux_bash.run scenario for it to work by changing the lines
   append_if [have_spec x86] config {
around the rtc driver start node and route into
   append_if [expr {[have_spec x86] && ![have_spec linux]}] config {
Otherwise it might crash with errors like:
[init -> rtc_drv] Error: IO_PORT-session creation failed (ram_quota=6K, io_port_base=112, io_port_size=2)
[init -> rtc_drv] Error: Uncaught exception of type 'Genode::Parent::Service_denied'

alex-ab, 2017-03-31:
> [init -> noux -> /bin/bash] pthread_main_np_int: pthread_main_np called, not implemented
> Then bash gives an error: A NULL argv[0] was passed through an exec system call.

Do you have the fix of #2312 already on your branch or without ?

> When I do the same on Linux, I get:
> Kernel: ipc_cap_data nullptr

See post on mailinglist from yesterday:
https://sourceforge.net/p/genode/mailman/message/35758548/

BorisMulder-CSL, 2017-03-31:
I just merged with staging, and it does not change anything.

alex-ab, 2017-03-31:
Ah yes, you can in run in noux no multi-threaded applications, so no pthread_* support.

BorisMulder-CSL, 2017-04-03:
Well, it does not run anything, including ls or cat, except for builtins (cd, pdw).

chelmuth, 2017-04-03:
I just ran `KERNEL=nova make run/noux_bash` and tested several command, e.g., `echo 123 | base64`, and it works at expected. I used the current staging 09c5a1e88ec without further changes. Also, I did not see the message you posted above. Are you sure you're using a clean repository and an unmodified Genode tool chain?

BorisMulder-CSL, 2017-04-03:
Testing it on a completely clean work directory solved it.
It seems I made a mistake and did not copy some of the changes from staging into my own feature branch.

closed 2017-04-03 by BorisMulder-CSL