genodelabs/genode #2364: intel_fb_drv: intel_dsi_init not implemented

jklmnn, 2017-04-03:
On an Intel Atom the driver fails to start.

```
[init -> intel_fb_controller] --- Framebuffer controller ---
[init] child "acpi_report_rom" announces service "Report"
[init] child "acpi_report_rom" announces service "ROM"
[init -> nitpicker_config] Warning: top-level node <xray> missing in input ROM xray
[init] child "timer" announces service "Timer"
[init -> intel_fb_controller] 
[init] child "config_rom" announces service "ROM"
[init -> config_rom] request for intel_fb_drv -> fb_drv.config
[init -> nitpicker_config] Warning: could not obtain input value for input xray_enabled
[init -> acpi_drv] Error: No valid ACPI RSDP structure found
[init -> config_rom] request for intel_fb_drv -> fb_drv.config
[init -> intel_fb_drv] --- intel framebuffer driver ---
[init -> config_rom] request for intel_fb_drv -> fb_drv.config
[init] child "platform_drv" announces service "Platform"
[init -> intel_fb_drv] INFO: i2c-core: driver [dummy] registered
[init -> intel_fb_drv] ioremap: mapped phys 0x48000000 (size 4194304) to virt 0x400000
[init -> platform_drv] Error: Device 0x0:0x0.0 is used by more than one driver - session 'intel_fb_drv -> '.
[init -> platform_drv] Error: Device 0x0:0x2.0 is used by more than one driver - session 'intel_fb_drv -> '.
[init -> intel_fb_drv] ioremap: return sub range phys 0x48200000 (size 2097152) to virt 0x600000
[init -> intel_fb_drv] INFO: [drm] Memory usable by graphics device = 2048M
[init -> intel_fb_drv] [drm:ERROR] render: timed out waiting for forcewake ack request.
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus ssc] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus vga] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus panel] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpc] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpb] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpd] registered
[init -> intel_fb_drv] INFO: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[init -> intel_fb_drv] INFO: [drm] Driver supports precise vblank timestamp query.
[init -> intel_fb_drv] INFO: [drm] failed to find VBIOS tables
[init -> platform_drv] 0:2.0 uses IRQ, vector 0xff, MSI 32bit capable, non-maskable
[init -> platform_drv] Error: signal handler got not registered - irq object unavailable
[init -> platform_drv] Error: expected to find IRQ proxy for IRQ 0xff
[init -> intel_fb_drv] Warning: I/O port(964) write failed
[init -> intel_fb_drv] Warning: I/O port(965) read failed
[init -> intel_fb_drv] Warning: I/O port(965) write failed
[init -> intel_fb_drv] dev_dbg: adapter [DPDDC-B] registered
[init -> intel_fb_drv] dev_dbg: adapter [DPDDC-C] registered
[init -> intel_fb_drv] intel_dsi_init not implemented
[init -> intel_fb_drv] BUG: failure at /.../genode/repos/dde_linux/src/drivers/framebuffer/intel/dummies.c:556/intel_dsi_init()!
```

After applying https://github.com/jklmnn/genode/commit/18bc17a834725cfbc42349de7f1f28e98c2e07ea the log looks as follows:

```
init -> intel_fb_drv] --- intel framebuffer driver ---
[init] child "platform_drv" announces service "Platform"
[init -> config_rom] request for intel_fb_drv -> fb_drv.config
[init -> intel_fb_drv] INFO: i2c-core: driver [dummy] registered
[init -> intel_fb_drv] ioremap: mapped phys 0x48000000 (size 4194304) to virt 0x400000
[init -> platform_drv] Error: Device 0x0:0x0.0 is used by more than one driver - session 'intel_fb_drv -> '.
[init -> platform_drv] Error: Device 0x0:0x2.0 is used by more than one driver - session 'intel_fb_drv -> '.
[init -> intel_fb_drv] ioremap: return sub range phys 0x48200000 (size 2097152) to virt 0x600000
[init -> intel_fb_drv] INFO: [drm] Memory usable by graphics device = 2048M
[init -> intel_fb_drv] [drm:ERROR] render: timed out waiting for forcewake ack request.
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus ssc] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus vga] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus panel] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpc] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpb] registered
[init -> intel_fb_drv] dev_dbg: adapter [i915 gmbus dpd] registered
[init -> intel_fb_drv] INFO: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[init -> intel_fb_drv] INFO: [drm] Driver supports precise vblank timestamp query.
[init -> intel_fb_drv] INFO: [drm] failed to find VBIOS tables
[init -> platform_drv] 0:2.0 uses IRQ, vector 0xff, MSI 32bit capable, non-maskable
[init -> platform_drv] Error: signal handler got not registered - irq object unavailable
[init -> platform_drv] Error: expected to find IRQ proxy for IRQ 0xff
[init -> intel_fb_drv] Warning: I/O port(964) write failed
[init -> intel_fb_drv] Warning: I/O port(965) read failed
[init -> intel_fb_drv] Warning: I/O port(965) write failed
[init -> intel_fb_drv] dev_dbg: adapter [DPDDC-B] registered
[init -> intel_fb_drv] dev_dbg: adapter [DPDDC-C] registered
[init -> intel_fb_drv] [drm:ERROR] crtc 27: Can't calculate constants, dotclock = 0!
[init -> intel_fb_drv] [drm_wait_one_vblank] *WARN* vblank wait timed out on crtc 0
[init -> intel_fb_drv]  
[init -> intel_fb_drv] [drm_wait_one_vblank] *WARN* vblank wait timed out on crtc 0
[init -> intel_fb_drv]  
[init -> intel_fb_drv] [drm_property_free_blob] WARN_ON(!mutex_is_locked(&blob->dev->mode_config.blob_lock)) 
Warning: Quota exceeded! amount=27173, size=4096, consumed=24576
Warning: could not allocate metadata
...
Warning: Quota exceeded! amount=133669, size=4096, consumed=131072
Warning: could not allocate metadata
[init -> intel_fb_drv] ioremap: mapped phys 0x40003000 (size 131072) to virt 0x360000
[init -> intel_fb_drv] ioremap: mapped phys 0x40027000 (size 131072) to virt 0x380000
[init -> intel_fb_drv] ioremap: mapped phys 0x4004a000 (size 131072) to virt 0x3a0000
[init -> intel_fb_drv] INFO: [drm] Initialized i915 1.6.0 20151010 for dummy on minor 0
[init -> intel_fb_drv] Warning: I/O port(964) write failed
[init -> intel_fb_drv] Warning: I/O port(965) read failed
[init -> intel_fb_drv] Warning: I/O port(965) write failed
[init -> intel_fb_drv] INFO: [drm] GMBUS [i915 gmbus vga] timed out, falling back to bit banging on pin 2
[init -> intel_fb_drv] Warning: I/O port(964) write failed
...
[init -> intel_fb_drv] Warning: I/O port(965) write failed
[init -> report_rom] report 'intel_fb_drv -> connectors'
[init -> report_rom]   <connectors>
[init -> report_rom]    <connector name="VGA-36" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-40" connected="false"/>
[init -> report_rom]    <connector name="DP-46" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-50" connected="false"/>
[init -> report_rom]    <connector name="DP-53" connected="false"/>
[init -> report_rom]   </connectors>
Warning: Quota exceeded! amount=59941, size=12288, consumed=49152
[init -> intel_fb_drv] ioremap: mapped phys 0x40080000 (size 4096) to virt 0x354000
[init] child "intel_fb_drv" announces service "Framebuffer"
[init -> report_rom] report 'intel_fb_drv -> connectors'
[init -> report_rom]   <connectors>
[init -> report_rom]    <connector name="VGA-36" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-40" connected="false"/>
[init -> report_rom]    <connector name="DP-46" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-50" connected="false"/>
[init -> report_rom]    <connector name="DP-53" connected="false"/>
[init -> report_rom]   </connectors>
[init -> intel_fb_drv] ioremap: mapped phys 0x400a0000 (size 4096) to virt 0x35b000
```

It seems that it doesn't recognize the screen. Would implementing `intel_dsi_init` fix this? I thought about using the implementation from https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/i915/intel_dsi.c (or maybe an older version, if required).

cnuke, 2017-04-03:
Judging by a brief look at the source I'm not sure that implementing only `intel_dsi_init` is enough to get the display working. So I would rather recommend to incorporate the whole file into `dde_linux`. I'm afraid there is no proper guide line on how to do that but it basically boils down to adding the contrib file to the `intel_fb.list` as well as adding it to `intel_fb_drv.mk` (although the library description already uses wildcards and should pick up the file automatically). After that it is just the usual “get to a point where the code compiles” game.

chelmuth, 2017-04-03:
I'm not that familiar with intel_fb_rv but what also puzzles in the log are the following lines
```
[init -> platform_drv] 0:2.0 uses IRQ, vector 0xff, MSI 32bit capable, non-maskable
[init -> platform_drv] Error: signal handler got not registered - irq object unavailable
[init -> platform_drv] Error: expected to find IRQ proxy for IRQ 0xff
```
Does the driver depend on IRQs or can this just be ignored?

cnuke, 2017-04-03:
The driver can work w/o interrupts (remember the polling “fix” for the X201 to detect hotplug events that normally raise an interrupt). That being said the error should be investigated.

jklmnn, 2017-04-12:
I have tracked it down to the VBIOS which is used (at least in Debian) to access the Display driver. The problem now is, the structure is created in `intel_opregion_setup` in [`i915_drv.h:3421`](http://lxr.free-electrons.com/source/drivers/gpu/drm/i915/i915_drv.h#L3421) wich is only implemented if `CONFIG_ACPI` is defined. The problem is that it gets implemented as static inline function if it is not defined. So I have no way to use it without either changing the kernel source or defining `CONFIG_ACPI` (which pulls more depencies). Is there any canonical way to solve this problem?

jklmnn, 2017-05-02:
I have implemented it to the point that it recognizes the VBIOS table. If someone wants to continue the implementation, it's located at https://github.com/jklmnn/genode/tree/patch_i915_drv

nfeske, 2017-06-06:
Is this being actively worked on or shall we better close the issue for now?

jklmnn, 2017-06-06:
From my side, this can be closed. My current work is linked to in my comment above if anyone wants to continue from this point.

nfeske, 2017-06-06:
Thanks for the quick response.

closed 2017-06-06 by nfeske