genodelabs/genode #2365: rump_fs gives read only access to file system if policy is absent

BorisMulder-CSL, 2017-04-03:
If I specify no policy in rump_fs for clients, it gives me read-only access (I expected it to throw service_denied).

You can verify this when you comment out the policy in rump_ext2.run:
```
<start name="rump_fs">
	<resource name="RAM" quantum="8M" />
	<provides><service name="File_system"/></provides>
	<config fs="ext2fs"><!--<policy label_prefix="test-libc_vfs" root="/" writeable="yes"/>--></config>
</start>
```
Now, the mkdir() test will fail with:
mkdir(dir_name, 0777) failed, ret=-1, errno=1
So instead of the File_system session failing, it gives EPERM.
In one of my scenarios, it gives noux/bash read-only access to the file system.
I guess this is a bug?

chelmuth, 2017-04-03:
Surprisingly, this behavior was intended by the original author that's documented in the source code https://github.com/genodelabs/genode/blob/master/repos/dde_rump/src/server/rump_fs/main.cc#L430
```c++
/*
 * If no policy matches the client gets
 * read-only access to the root.
 */
```
However, I admit that this is not expected from a component supporting the policy configuration. Thanks for reporting, I'll mark this issue as cleanup.

BorisMulder-CSL, 2017-04-03:
> by the original author
https://github.com/genodelabs/genode/blob/master/repos/dde_rump/src/server/rump_fs/main.cc#L3
I see what you did there ;)

ssumpf, 2017-04-03:
@BorisMulder-CSL:
> by the original author
... would be neither of the two in the header in this case.

chelmuth, 2017-04-03:
Hm, seems @ehmry changed the policy handling in 632444ae232a6f1b7c814d900e4f9e5e46103f84 mid-2016 and I vaguely remember there was some rationale...

ehmry, 2017-04-03:
I can't think of one, this was after ``default-policy``. At least I documented the bad behavior :-) 

I volunteer to fix this and check the other servers.

ehmry, 2017-04-03:
Fixed at 4dc61c3474be2aeed052c40233eed2986ec8dfe8. I left the root offset policies as they are, ``ram_fs`` will deny clients without a root policy and ``rump_fs`` and ``vfs`` will default to no offset. However ``ram_fs`` has been amended to apply a second client specified offset, consistent with the other two servers.

ehmry, 2017-04-05:
@chelmuth fixup at 6af215a24ff156abce3438e4d2e8f3dfc3439392

chelmuth, 2017-04-05:
@ehmry does this change something beyond b228117d34865e3933fdf0d230904d37e83a3a3c?

ehmry, 2017-04-05:
Yes, changes the exceptions that are thrown and removes a superfluous try block.

chelmuth, 2017-04-05:
Please rebase on current staging / my fixup commit b228117d34865e3933fdf0d230904d37e83a3a3c

ehmry, 2017-04-05:
Ok, just 1b414ff27d6a944a31d54c36893969040255fcd1 then.

chelmuth, 2017-04-05:
@ehmry many thanks, merged it to staging.

commit 5af3290c (2017-04-04): Unify client policy across File_system servers

File_system servers shall deny clients not matching a defined policy.
Servers shall also apply session root offset policy followed by a client
offset.

Fix #2365
commit f134c587 (2017-05-02): Unify client policy across File_system servers

File_system servers shall deny clients not matching a defined policy.
Servers shall also apply session root offset policy followed by a client
offset.

Fix #2365
closed 2017-05-31 by chelmuth