genodelabs/genode #2366: wm gives error when starting and killing subsystems from cli_monitor several times

BorisMulder-CSL, 2017-04-04:
Whenever I start and stop some subsystem in cli_monitor several times (about 3 times in my case), I get the following error:

[init -> wm] Error: could not look up server object, return from call id_pt=34957
[init -> cli_monitor -> bla] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> cli_monitor -> bla] Warning: abort called - thread: ep
[init -> cli_monitor] subsystem "bla" exited with value 1
no RM attachment (READ pf_addr=0x28 pf_ip=0x71f42 from pager_object: pd='init -> wm' thread='ep')
page fault, pd='init -> wm' thread='ep' cpu=0 ip=0x71f42 address=0x28 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1

After, the entire system hangs. However, the first two times I type "start bla" and "kill bla" in the cli, it just works.
In this case, bla is just a backdrop component routed to wm which only displays an image. The same happens with other subsystems.
Any idea what causes this?

BorisMulder-CSL, 2017-04-04:
I checked it again and apparently it goes wrong the first time any subsystem is restarted (after being started and stopped the first time). For instance, my noux subsystem does the following from cli_monitor:
> start noux [everything OK]
> kill noux [stopped noux]
> start noux [now I see a window with noux, but I cannot give any input]
> kill noux [now the frame of the window does not go away, although the subsystem no longer exists]
> start noux [nothing happens]
> start noux [now it starts a second instance of noux, noux.2, which works normally]
> kill noux [now the system hangs]

nfeske, 2017-04-05:
As far as I can tell, this symptom is not known so far. @BorisMulder-CSL Would you mind posting a link to a branch with the scenario, so that the issue can be easily reproduced?

The problem seems to lie within the `wm`. The message "Error: could not look up server object, return from call id_pt=34957" originates from the NOVA-specific base library. I wonder, does a similar message occur when starting the scenario on one of the other kernels?

BorisMulder-CSL, 2017-04-06:
On linux it gives me the following error whenever I try to start a subsystem that I've just killed:

[init -> wm] Error: inserting element 0x2b8676afa868 twice into avl tree!

BTW my own scenario is a bit complex, so I tried looking for other run scripts that use cli_monitor. However, the only one I can find is terminal_mux and that one does not run at all.
When I run it, it fails to start qemu:

spawn qemu-system-x86_64 -no-kvm -display sdl -cpu core2duo -m 256 -nographic -serial file:kdb.log -serial mon:stdio -cdrom var/run/terminal_mux.iso
[[73;124R
and it hangs.

BorisMulder-CSL, 2017-04-06:
My config of cli_monitor looks somewhat like this:
```

	<start name="wm" priority="-1">
		<resource name="RAM" quantum="12M"/>
		<provides>
			<service name="Nitpicker"/>
		</provides>
		<config>
			<policy label_prefix="decorator" role="decorator"/>
			<policy label_prefix="launcher -> menu"    role="direct"/>
			<policy label_prefix="layouter"  role="layouter"/>
		</config>
		<route>
			<service name="ROM" label="resize_request"> <child name="wm_report_rom"/> </service>
			<service name="ROM" label="focus"> <child name="wm_report_rom"/> </service>
			<service name="Report"> <child name="wm_report_rom"/> </service>
			<service name="Nitpicker"> <child name="nitpicker"/> </service>
			<any-service> <parent/> <any-child/> </any-service>
		</route>
	</start>

        <start name="cli_nit_fb" priority="-1">
		<binary name="nit_fb"/>
		<resource name="RAM" quantum="8M" />
		<provides>
			<service name="Framebuffer" />
			<service name="Input" />
		</provides>
		<config width="640" height="480" />
		<route>
			<any-service>
				<child name="wm"/> <parent/> <any-child/>
			</any-service>
		</route>
	</start>

	<start name="cli_terminal" priority="-1">
		<binary name="terminal"/>
		<resource name="RAM" quantum="2M"/>
		<provides><service name="Terminal"/></provides>
		<config>
			<keyboard layout="us"/>
			<font size="12" />
		</config>
		<route>
			<any-service>
				<child name="cli_nit_fb"/> <parent/> <any-child/>
			</any-service>
		</route>
	</start>

	<start name="cli_monitor" priority="-3">
		<resource name="RAM" quantum="2020M"/>
		<config verbose="yes">
			<vfs>
				<dir name="subsystems">
}

set subsystem_files {
	bla.inc
	noux.inc

}
#	windows.inc
#	linux.inc

# source include files 
foreach file $subsystem_files { 	
	source ${genode_dir}/repos/mydir/$file
} 

append config {
				</dir>
			</vfs>
		</config>
		<route>
                        <service name="ROM" label="focus">
				<child name="wm_report_rom" />
			</service>
			<service name="File_system" label="config">
				<child name="config_fs"/>
			</service>
			<any-service>
				<child name="cli_terminal"/>
				<child name="wm"/>
				<parent/> <any-child/>
			</any-service>
		</route>
	</start>
```
If you need to see more, let me know.

nfeske, 2017-12-13:
I was able to reproduce the problem. It only occurred with the themed decorator, but not the default decorator. The themed decorator creates/destroys nitpicker sessions for each window, which does not happen for the default one. The code for closing a decorator's nitpicker session was missing a desolve call. The problem is fixed on staging via 0e4d676d7ce34be2ba3ca90da5577eed8b3219a4.

chelmuth, 2017-12-21:
Fixed in master by bf60e1486d9dc58d87546e08e668f682e7a5211d.

closed 2017-12-21 by chelmuth