genodelabs/genode #2369: Register framework: the warning, the spacing, and the foreigner

m-stein, 2017-04-05:


m-stein, 2017-04-05:
@cnuke You may want to give 1320960 a try ;)

@chelmuth Please have a look at these commits:
1320960 Register_set: use Genode::Registers as interface
cba479a Register: avoid warnings about oversized shifts
81b36a8 Register_array: introduce SPACING parameter

Unfortunately, in1320960, I have found no other way but using 2 template parameters in Register_set::read/write (seems impossible in C++ without bloating it too much).

cnuke, 2017-04-05:
@m-stein Alright, will do — thank you very much!

nfeske, 2017-04-06:
I sense that we are risking to go overboard with the register framework. It is already more complex than I would like it to be and the latest patches by @m-stein follow this direction further. Instead of extending it, I would strongly advocate trimming it.

Of course I don't know all the drivers, but I have the impression that some features, in particular the register array, could be alleviated by just using `Mmio` objects at a finer granularity. E.g., if there is a repeated register block, we could model the register block at a distinct `Mmio` that can naturally be instantiated several times. I.e., combined with `Constructible`, such fine-grained `Mmio` objects could be kept in a regular array - presumably covering the use case of the register array.

@m-stein I would greatly appreciate you to review the current uses of the the advanced features of the register framework. In particular, I would like to get an assessment about the possible removal of the register array, by using fine-grained `Mmio` objects instead. Right now, I feel that the register array already defeats the original intention of the register framework to make _things obvious_ while still being not flexible enough (as evidenced by your 'SPACING' patch, or the fact that it takes four numbers as template arguments).

m-stein, 2017-04-06:
@nfeske That's a cool idea with the arrays, indeed. Haven't thought about that yet. I'll give it a try. I absolutely agree that it would be nice to reduce the complexity of the framework. IMO, there was already a step into this direction with the new wait-for implementation. It's not that I like accumulating more and more template-ism, so I'm always open for new ideas ;)

m-stein, 2017-04-06:
@nfeske Btw. these commits are merely suggestions and of course not set in stone. I appreciate the reflections and discussions they bring-up even more than their current state.

m-stein, 2017-04-06:
After some further offline discussion, we decided to merge only
3bfb681 Register: avoid warnings about oversized shifts

AFAIK, the interface problem can't be solved in C++11 without bloating the interface. We should stay with the solution of partially specialized templates by now. The spacing problem can be solved with the suggested ``Mmio``-array approach. This way, we spare another irritating parameter at the register array.

chelmuth, 2017-05-02:
Unfortunately, 3bfb681f57ae39b4184f8102df9fc4c478b4297f does no longer apply. Please rebase on current staging.

chelmuth, 2017-05-31:
Still open, but fixed?

closed 2017-05-31 by m-stein