genodelabs/genode #2373: Nic_router: Session_component::link_state not implemented

BorisMulder-CSL, 2017-04-07:
When I try to connect virtualbox to the nic router, it gives this error. Am I missing this implementation somewhere, or does it not exist? I cannot do any networking through vbox with nic_router this way, even if I change the unimplemented function to return true.

BorisMulder-CSL, 2017-07-21:
I have looked at this again, and the following behaviour seems to happen:
- When I use Virtualbox with a Nic connection through a nic_bridge and with dhcp, it works fine.
- When I use Virtualbox with a nic_router and with dhcp, it does not work.
- When I use Virtualbox with a nic_bridge and with static ip, it does not work.
- When I use Virtualbox with a nic_router and with static ip, it does not work.
- When I use another application (in this case vfs-lxip with some udp client) in the same scenario, it works with nic_router, nic_bridge, dhcp and static ip.

So it seems to me that the following conclusions can be drawn:
- Virtualbox does not support static IP (or I did something wrong and because of that static IP does not work).
- The nic_router does not support DHCP (or I did something wrong again).

If I did anything wrong, what needs to be configured for any of the above combinations to work?
If it is indeed not supported, will it be supported in the future?

m-stein, 2017-08-07:
Hi Boris,
Sorry for the delayed answer - I was on vacations. The NIC router does not inspect a packets DHCP information. The headers that are inspected are the following:
```
Eth --+--> ARP
      |
      +--> IPv4 --+--> TCP
                  |
                  +--> UDP
```
So technically speaking, the routing of DHCP packets normally follows your UDP settings at the router. What you trying to do is configuring a network behind one router interface with a DHCP server that is located behind another interface. This isn't possible. The router is not meant to be used with IP subnets that span over different interfaces. Thus, there is no way to forward broadcasts for example. This is different with the NIC bridge that has no notion of IP subnets.

We had some internal discussion regarding DHCP and our goal is to write an extra server component for this. This server takes the subnet configuration as config input and then interacts with the router. It shall also reconfigure the router according to the changing subnet states. I plan to start implementing it like in the next months.

BorisMulder-CSL, 2017-08-08:
Ok, this would probably help.
Regarding my other point, is there a way to configure virtualbox to use a static IP?

alex-ab, 2017-08-08:
We never tried with static IP addresses, so you are on unexplored ground.

MartijnVerschoor, 2017-08-11:
I welcome the development of a DHCP component. I do wonder if it brings a significant value over statically assigning IP addresses for local networks behind the nic_router for the typical use cases. The IP addresses used on this local network are - well - local to this network and hence never interfere with the external (upstream) network. Thus there is also no interference with multiple instances of the same integrated Genode product. The value for DHCP on such local network lies in 1) not requiring to configure an IP address at all and 2) dynamic scenarios with multiple instances of a component with the same configuration. Also, I wonder if in these cases the value of not having to configure an IP address would not lead to configuring unique MAC addresses ;-)

I would surely not dismiss the support for static IP addresses in favor of DHCP.

@alex-ab, thanks for the heads-up. What do you think, would it be less work to fix static IP addresses for VirtualBox compared to implementing a DHCP server component?

alex-ab, 2017-08-14:
I can't give you a good answer. The static IP address setup should work with nic_bridge, but it does not. Maybe it is just a matter of debugging one day or so (but who knows how long it takes). The DHCP server component approach is of course more development effort, but effectively is more flexible (compared to manually assigning and maintaining static IP addresses in nic_bridge & in all VMs in all kind of scenarios).

m-stein, 2017-09-06:
@BorisMulder-CSL @MartijnVerschoor I gave static IP addresses + vbox + tinycore + router a try on my branch [2490_virtualbox_network_access_through_nic_router](https://github.com/m-stein/genode/tree/2490_virtualbox_network_access_through_nic_router) and fixed some problems (e.g. support for link states) that could have been a show stopper in your setup too. I then did the next step and implemented a simple DHCP server in the NIC router. An example that I tested with a TinyCore guest in Virtualbox can be found in ports/run/virtualbox_nic_router.run on the branch.

The DHCP server can be activated for an interface through the config of the NIC router:
```
	<domain name="my_virt_net" interface="10.0.1.1/24">
		<dhcp-server ip_first="10.0.1.80"
		             ip_last="10.0.1.100"
		             ip_lease_time="3600"
		             dns_server="10.0.4.3"/>
		<tcp dst="0.0.0.0/0"><permit-any domain="uplink"/></tcp>
		<udp dst="0.0.0.0/0"><permit-any domain="uplink"/></udp>
	</domain>
```

IP first and last define the range of IP addresses from which the server can lend addresses to clients. IP lease time is the timeout for loans in seconds. DNS server is optional and might contain a local address or an address from a foreign subnet. The server additionally sends its own IP address (interface attribute of the domain) in the DHCP fields "router IP" and "server IP". It also adds "broadcast address" and "subnet address" values according to the interface attribute of the domain.

Current restrictions:
The lease time must be under 4294. The server only handles DHCP discover and request messages and for instance no refresh messages. It nonetheless releases IP allocations if their lease time expires. Thus, it might be best to always use the maximum lease time and IPs can't be released by the client.

I'm open for suggestions. The plan is to get the current state clean with full support of DHCP message types and then merge it to mainline. Afterwards we might move the functionality to an extra component as depicted above.

MartijnVerschoor, 2017-09-08:
@m-stein, I tried your tinycore scenario with both DHCP and static IP. I can confirm that both work. So far I did not encounter any problems, thus it is very encouraging! Also, I like your tinycore setup. It is very fit for these kind of test scenarios.

Next step for us is integration in our scenario, test with other guest OSs and stress it a bit more. Ideally after it is merged to main.

Thanks!

m-stein, 2017-09-08:
@MartijnVerschoor The implementation on my branch now suports DHCP fully. I'll clean it up and think it should be merged during next week. Thanks for testing it ;)

MartijnVerschoor, 2017-09-25:
@m-stein, we are ready to give the modified nic_router another spin. I still see some transients on the related issue #2490, but those appear to be deviating from the original topic. Could you inform us if it is prompt to proceed and, if so, which commits to pick?

m-stein, 2017-09-25:
@MartijnVerschoor I'm currently in the clean-up/rebasing process. I try to get a good state for you to test today and will then inform you about it.

m-stein, 2017-09-25:
@MartijnVerschoor My merge_to_staging branch contains a clean state that is based on current staging. I hope it does work fine for you?

MartijnVerschoor, 2017-09-28:
@m-stein, thanks! @BorisMulder-CSL and I performed a test-integration in our scenario. We cherry-picked [these commits](https://github.com/genodelabs/genode/issues/2490#issuecomment-331948887), and these as well:
2853339f68a10817f650ca7cdcac99aa350a24a5
682bb9a372bbd337735865a4e4ae032905809379
8cd91dd6c9390bcc5433acd74230139d73273c37
f5d7e48db484196053a7f3b3a94c81548325e257
bf0988b6c487535c4280db46531db61ed29f3418
59709cf5eb16de241270fa462e9af3a511ef0f49
144d4215d836e53192bc1887fb47662c2a88ae60
82c763fe75e33cf012a2ac3093396fc0a482af69
a1e32a3ad97cea55b0c463f016f085584fbd8f89
68cd5e80041ca716d4e63a657d70e5b51831ee13

on top op 17.08.

We tested with a Windows 8.1 64 bit guest with DHCP provided by the router. The router configuration is almost similar to the one provided in your virtualbox_nic_router.run script.

A short report of our first findings:

1. It takes one to two minutes before Windows gets an IP address

During that period we see ~7 bad DHCP requests:

`[init -> nic_router] Error: bad DHCP request`

I did not (yet) investigate this further.

After that period Windows gets handed an IP address and we can access the network and the Internet.

2. The nic_router aborts when stopping/killing VirtualBox:

... sometimes with this message:
`[init -> nic_router] Error: Uncaught exception of type 'Genode::Bit_array_base::Invalid_clear'`
`[init -> nic_router] Warning: abort called - thread: ep`

... and sometimes with this message:

`[init -> nic_router] Warning: waiting for ARP cancelled`
`[init -> nic_router] Error: __cxa_pure_virtual called, return addr is 0x101c1f9`
`[init -> nic_router] Warning: abort called - thread: ep`

3. We also tested the `vfs` server with lxip. Configured with static ip_address it works fine. We did not (yet) managed to get it working with DHCP (using \<lxip dhcp="yes"/\>).

`[init -> vfs] Using DHCP for interface configuration.`
`[init -> nic_router] (router <- vfs) ETH 02:02:02:02:02:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 1 02:02:02:02:02:00 > 0.0.0.0`
`[init -> nic_router] Error: bad DHCP request`
`[init -> nic_router] (router <- uplink) ETH 14:18:77:69:27:d5 > 50:7b:9d:3e:a6:57 ARP 1 14:18:77:69:27:d5 10.0.0.2 > 10.0.0.116`
`[init -> nic_router] Ignore ARP request`

4. With the above commits, on some machines our scenario stalls with the observation that the `nic_drv` does not come to announce its Nic service... Your virtualbox_nic_router scenario runs fine on all those machines. It may point to a weak spot in our scenario so we want to investigate what may have caused it. Do you think [this](https://github.com/genodelabs/genode/issues/2490#issuecomment-332508822)? could cause any trouble?

m-stein, 2017-09-28:
Hi Martijn,

Thanks for your fast and detailed feedback! :)

> A short report of our first findings:
>
>  1. It takes one to two minutes before Windows gets an IP address
>
> During that period we see ~7 bad DHCP requests:
>
> |[init -> nic_router] Error: bad DHCP [...]

I was able to reproduce the "bad DHCP requests" and "Bit_array_base::Invalid_clear" with vbox5_win7_64.run (haven't tested win8 so far) and will now investigate further on these issues.

> 3. We also tested the |vfs| server with lxip. Configured with static ip_address it works fine. We did not (yet) managed to get it working with DHCP (using <lxip dhcp="yes"/>).
>
> |[init -> vfs] Using [...]

I hope this gets solved while fixing the win7 scenario. Otherwise I'll have a more detailed look at this afterwards.

> 4. With the above commits, on some machines our scenario stalls with the observation that the |nic_drv| does not come to announce its Nic service... Your virtualbox_nic_router scenario runs fine on [...]

I don't think so but to be sure: I've changed the commits on my merge_to_staging branch in the
meantime to solve the problem that Christian observed. These commits are relevant for the timer
problem if you want to give it a try:

089c76b timer: limit rate of handling timeouts
f42da0a timer test: absence of starvation under stress
938713f signal: fix starvation by fast signal contexts

As soon as I have fixes for the WinVM problems I'll give you a hint.

m-stein, 2017-09-29:
With the current state of my merge_to_staging branch the Vbox Guest gets its IP on the first DHCP exchange, there are no "bad DHCP packets", and the NIC router always survived the Vbox exit. I tested it with these run scripts:
vbox5_win7_64
vbox4_win81_64
vbox5_win10_64

For the VFS+LXIP use-case I used the netty_udp/tcp run scripts and saw that at least DHCP is now succesful and free from "bad packets".

However, I could not trigger this:
```
[init -> nic_router] Warning: waiting for ARP cancelled
[init -> nic_router] Error: __cxa_pure_virtual called, return addr is 0x101c1f9
[init -> nic_router] Warning: abort called - thread: ep
```
Maybe you could lookup the reported return address and post the result to bring some more light into this? Also, the verbose nic_router log could be useful here.

I left my temporary commits with the adapted the run scripts on the branch, so you may compare them to your own integration setup.

m-stein, 2017-11-10:
As the NIC router now acts as DHCP server as well as DHCP client and we successfully use it in some Virtualbox scenarios, may I close the issue?

BorisMulder-CSL, 2017-11-13:
We have to test it yet, but for now you can close it. If we find anything else, we will submit a new issue.

m-stein, 2017-11-13:
Ok, thanks.

closed 2017-11-13 by m-stein

nfeske, 2017-11-13:
I am using it right now for posting this comment. It works nicely! :-)

m-stein, 2017-11-13:
Nice :)