genodelabs/genode #2379: NOVA: flexible array member in union

cproc, 2017-04-10:
When building NOVA with GCC 6, compile errors occur:

```
In file included from /.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/src/acpi.cpp:28:0:
/.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/acpi_rsdt.hpp:45:25: error: flexible array member in union
             uint32  rsdt[];
                          ^
/.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/acpi_rsdt.hpp:46:25: error: flexible array member in union
             uint64  xsdt[];
                          ^
```

```
In file included from /.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/ec.hpp:30:0,
                 from /.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/src/bootstrap.cpp:23:
/.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/regs.hpp:57:23: error: flexible array member in union
             mword gpr[];
                       ^
```

```
In file included from /.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/svm.hpp:23:0,
                 from /.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/src/cpu.cpp:34:
/.../contrib/nova-2c8890618fa5f5db286c7efae4736c4aa6d5061f/src/kernel/nova/include/utcb.hpp:93:22: error: flexible array member in union
             mword mr[];
                      ^
```

Ref #2372

nfeske, 2017-04-11:
I remember that @cproc already discussed this problem with me. The most straight-forward solution I can think of is to introduce a `NUM_GPRS` (enum or #define) definition in a header file that differentiates 32 and 64 bit. Instead of using an empty array, we dimension the array with `NUM_GPRS`. To make sure that the array is indeed larger or equal to the size of the structured variant, we should place a `static_assert` just beyond the union definition. This approach loosely resembles the one that is extensively used in the VirtualBox code base.

Another idea would be to not use a union but to introduce accessor methods for reading/writing the Nth register. The accessor methods could cast `this` to a `mword_t` and add the N to access the requested field. But I am afraid that this variant would result in a quite elaborate patch as it would change the indentation of the current now nested union/struct members.

alex-ab, 2017-04-12:
I will take care, but actually there was a patch 5377940bcb98d1491e5e59d2c7cadfdf7f6634e7.

cproc, 2017-04-12:
The linked patch causes warnings about array access above bounds and was just a quick hack to get NOVA running again.

alex-ab, 2017-04-12:
@chelmuth: please consider d1b75ef18c95379c5d8fe5f6360edc996d142d28 for staging

nfeske, 2017-04-12:
@alex-ab Your NOVA changes in https://github.com/alex-ab/NOVA/blob/74ba9b561f3a635e313447c63cf0b9241cbef560/include/regs.hpp look very sensible. But I am wonderung about the dimensioning of the `gpr` array:

```
mword gpr[sizeof(mword) * 2];
```

So on x86_32, there are 64 general-purpose registers, and there are twice as much on x86_64. It is not obvious where this particular upper bound comes from, and how it relates to the actual hardware.

alex-ab, 2017-04-12:
32bit: sizeof(mword) is 4 * 2 = 8 gpr register
64bit: sizeof(mword) is 8 * 2 = 16 gpr register

chelmuth, 2017-04-12:
Correct, but isn't this just coincidence? I'd prefer an explicit `if x86_64` of some kind.

nfeske, 2017-04-12:
@alex-ab Thanks for getting me on the right track regarding `sizeof`. ;-) Of course I should blame myself about my wrong conclusion. The use of a named constant or a comment would have avoided my misinterpretation in the first place, though.

alex-ab, 2017-04-12:
> Correct, but isn't this just coincidence?

Nope, intention.

> I'd prefer an explicit if x86_64 of some kind.

I tried really hard, but - no. Ifdef's are sooo ugly and I have to add sooo much lines of code - no - sorry.

(I really could be used to that ... )

> The use of a named constant or a comment would have avoided my misinterpretation in the first place, though.

Sure, that is the Genode way.

Maybe as compromise, what about:

```
mword gpr[(sizeof(mword) * 8 == 32) ? 8 : 16];
```

which is potentially too verbose already ...
(I'm going to keep down)

chelmuth, 2017-04-12:
Looks better indeed. Merged it although it has some sour taste ;-)

commit 71b11298 (2017-04-12): nova: fix compiler warnings/errors with GCC 6

Fixes #2379
commit 2da56551 (2017-04-12): nova: build fixes with GCC 6 of kernel

Fixes #2379
Fixes #2380
closed 2017-05-31 by chelmuth