genodelabs/genode #2381: rom_filter: compile error with GCC 6

cproc, 2017-04-10:
When building the 'rom_filter' component with GCC 6, a compile error occurs:

```
In file included from /.../repos/os/src/server/rom_filter/main.cc:24:0:
/.../repos/os/src/server/rom_filter/input_rom_registry.h: In lambda function:
/.../repos/os/src/server/rom_filter/input_rom_registry.h:360:50: error: passing ‘const Rom_filter::Input_rom_registry’ as ‘this’ argument discards qualifiers [-fpermissive]
     input_value = _query_value_in_roms(input_node);
                                                  ^
/.../repos/os/src/server/rom_filter/input_rom_registry.h:285:15: note:   in call to ‘Rom_filter::Input_value Rom_filter::Input_rom_registry::_query_value_in_roms(Genode::Xml_node)’
   Input_value _query_value_in_roms(Xml_node input_node)
               ^~~~~~~~~~~~~~~~~~~~
```

Ref #2372

chelmuth, 2017-04-12:
Fixed this with 28fd3e41b0c7c9444fda4330457d000aa2c650c8. Did I miss your patch?

cproc, 2017-04-12:
@chelmuth: no, I had only a quick fix to be able to continue the build process and find out if more errors occur (https://github.com/cproc/genode/commit/dcacfa17610c444518a321fa495ec3be72889daf)

closed 2017-05-31 by cproc