genodelabs/genode #2388: hw: separate bootstrap and core strictly

skalk, 2017-04-11:
For strict separation and more clarity we have to separate core and bootstrap. Therefore, all code that is shared should reside at a common place, like src/lib/hw.

chelmuth, 2017-04-11:
Beside my single comment I'm for a merge as soon as possible.

skalk, 2017-04-12:
@chelmuth : I've added two fixup commits, see above, for easier review. The one fixes all mistakenly rebased old licence headers, the second fixes a problem for Cortex A9 that was  unfortunately overlooked by me, when changing the last platforms.

chelmuth, 2017-04-12:
Thanks, merged into staging.

chelmuth, 2017-04-19:
It seems the timer on imx53 is not working anymore. I suspect the commits in this issue but did not check thoroughly. Could you please have a look?

skalk, 2017-04-24:
@chelmuth here is a fixup eee42a035ef53173b7c8c2c33445317d2b121da0, actually not for the commit introducing the problem, but a fixup for the original commit bites the one that follows up. That's why the fixup addresses the follow-up commit.

skalk, 2017-06-01:
Fixed in release 17.05

closed 2017-06-01 by skalk