genodelabs/genode #2390: Fiasco.OC: build errors with GCC 6

cproc, 2017-04-11:
When building our current version of Fiasco.OC with GCC 6, errors occur:

```
      [sigma0]  /.../contrib/foc-04471082e6f2db5a80e190468691b73a68e2e206/src/kernel/foc/l4/pkg/sigma0/server/src/io_backend.cc:21: undefined reference to `operator delete(void*, unsigned long)'
```

```
      [fiasco.oc]  libc.a(puts.o): In function `memcpy':
      [fiasco.oc]  /.../contrib/foc-04471082e6f2db5a80e190468691b73a68e2e206/src/kernel/foc/kernel/fiasco/src/lib/minilibc/ia32/include/memcpy.h:10: multiple definition of `memcpy'
      [fiasco.oc]  libc.a(vprintf.o):/.../contrib/foc-04471082e6f2db5a80e190468691b73a68e2e206/src/kernel/foc/kernel/fiasco/src/lib/minilibc/ia32/include/memcpy.h:10: first defined here
```

```
  ==> Post-processing bootstrap.elf

ERROR: Missing at least one platform instantiation.
```

cproc, 2017-04-11:
I have a fix for the build errors in https://github.com/cproc/foc/commit/2d27cb7e869b46c8d3c66bcc6cbaaa530f4c8528, but there's still a runtime problem on the x86_32 platform. The 'run/log' test works fine on x86_64 and pxba9, but hangs very early on x86_32.

chelmuth, 2017-04-11:
The question for me is now: Should we change *foc.port* to refer to your repo for the time being?

commit 65750b59 (2017-04-12): foc: fix port build with GCC 6

Fixes #2390
closed 2017-05-31 by chelmuth