genodelabs/genode #2391: Init: state report: closed sessions of killed clients remain reported?

MartijnVerschoor, 2017-04-11:
(as discussed yesterday with @nfeske) While experimenting with the new init component together with @BorisMulder-CSL, we stumbled upon something of which we are not certain if it is intended.

For our experiments we created a simple test scenario based on the components of hello_tutorial. A dynamic_rom is used to alternate (every two seconds) the config of init to have either a hello_client and a hello_server, or only the hello_server. As expected the client starts and stops every period, the server remains running.

We were interested to see how sessions would be reported in the provided and requested sections. But since the (modern) hello_client immediately exits after one test iteration we decided to add an infinite loop (see patch below).

Now, the sessions are reported nicely but we noticed that the state of sessions keeps being reported long after they have been closed (see log). Ultimately the report generator runs out of space yielding the "Error: state report exceeds maximum size". Of course in this scenario the hello_client is killed and the session is not closed as graceful as it would have been otherwise, but we wonder if this behavior is intended.

A code inspection reveals that the report is generated based on the population of the server Id_space. Could this scenario eventually lead to exhaustion of the Id_space?

Test scenario: [init_test.run.txt](https://github.com/genodelabs/genode/files/913259/init_test.run.txt)
Log: [init_test.run.txt](https://github.com/genodelabs/genode/files/913396/init_test.run.txt)
hello_client patch: [patch.txt](https://github.com/genodelabs/genode/files/913399/patch.txt)

nfeske, 2017-04-12:
Thank your very much for the detailed report and the test. With your run script, I can perfectly reproduce the problem.

nfeske, 2017-04-12:
The commit https://github.com/genodelabs/genode/commit/a70d3dd13cc8615ebb3b0b361fa7299e05248c16 fixes the issue and extends the init test accordingly. Thanks again for pointing us to the problem!

nfeske, 2017-04-12:
Btw, just as a minor side note about your run script, the report-ROM server has a `verbose` config attribute, which is a quite convenient alternative to the `rom_logger`.

MartijnVerschoor, 2017-04-13:
Ok, thanks for the clarification and the commit. I'd like to report back, but I'm afraid that this commit pulls in some other work that requires a newer toolchain. Make reports:

    Makefile:128: *** "/usr/local/genode-gcc/bin/genode-x86-g++ version 6.3.0 is required".  Stop.

MartijnVerschoor, 2017-04-13:
At @nfeske, thanks for your tip about the `verbose` config attribute. I was already aware of this, but have also grown accustomed to use a separate `rom_logger`. I agree that in this case it would be more convenient. Often, in scenarios with more complex `report_rom` policies, I have found the `verbose` attribute to quickly produce too much noise. In these cases I use a separate `rom_logger` to filter only a single policy.

In fact, I already considered requesting a feature that would allow to set verbosity per `rom_report` policy. (at least if I'm not mistaken this is not yet supported right?)

chelmuth, 2017-04-13:
@MartijnVerschoor you're right, we're currently enabling the new tool chain on staging. You may try to just `git cherry-pick` @nfeske's commit.

MartijnVerschoor, 2017-04-13:
I can confirm the fix works for me. Thanks!

closed 2017-04-13 by MartijnVerschoor

nfeske, 2017-04-18:
> In fact, I already considered requesting a feature that would allow to set verbosity per rom_report policy. (at least if I'm not mistaken this is not yet supported right?)

As of now, the policies of the report-rom server refer to the component's ROM service. There are no policies for the report service. Hence, configuring a per-report verbosity is not easily possible without extending the configuration concept in some way. I am a bit hesitant.

Another possibility to limit the noise would be the use of multiple report-ROM instances for different purposes. Like one instance for the GUI, another one for the drivers, etc., where the verbosity of each instance can be toggled independently from the others.

MartijnVerschoor, 2017-04-18:
@nfeske, my consideration for `report_rom` verbosity policies per report are to provide even more convenience. As stated, we can achieve the same in other ways - admittedly not too complicated either. So I rest my case ;-) 

Thanks for the rationale.

commit a70d3dd1 (2017-04-12): base: drop session states of vanished clients

For asynchronously provided sessions, the parent has to maintain the
session state as long as the server hasn't explicitly responded to a
close request. For this reason, the lifetime of such session states is
bound to the server, not the client.

When the server responds to a close request, the session state gets
freed. The 'session_response' implementation does not immediately
destroy the session state but delegates the destruction to a client-side
callback, which thereby also notifies the client. However, the code did
not consider the case where [... cut]
commit 30871b72 (2017-04-12): base: drop session states of vanished clients

For asynchronously provided sessions, the parent has to maintain the
session state as long as the server hasn't explicitly responded to a
close request. For this reason, the lifetime of such session states is
bound to the server, not the client.

When the server responds to a close request, the session state gets
freed. The 'session_response' implementation does not immediately
destroy the session state but delegates the destruction to a client-side
callback, which thereby also notifies the client. However, the code did
not consider the case where [... cut]
commit 0a973a45 (2017-05-09): base: drop session states of vanished clients

For asynchronously provided sessions, the parent has to maintain the
session state as long as the server hasn't explicitly responded to a
close request. For this reason, the lifetime of such session states is
bound to the server, not the client.

When the server responds to a close request, the session state gets
freed. The 'session_response' implementation does not immediately
destroy the session state but delegates the destruction to a client-side
callback, which thereby also notifies the client. However, the code did
not consider the case where [... cut]
commit b8ad8023 (2017-05-09): base: drop session states of vanished clients

For asynchronously provided sessions, the parent has to maintain the
session state as long as the server hasn't explicitly responded to a
close request. For this reason, the lifetime of such session states is
bound to the server, not the client.

When the server responds to a close request, the session state gets
freed. The 'session_response' implementation does not immediately
destroy the session state but delegates the destruction to a client-side
callback, which thereby also notifies the client. However, the code did
not consider the case where [... cut]