genodelabs/genode #2395: VFS File_system plugin: inconsistent requirement for a leading path slash

ehmry, 2017-04-13:
All VFS plugins use absolute paths, but most support the omission of a leading path slash. The File_system plugin supports this for some but not all methods.

I prefer to keep leading slashes optional, just as DNS leaves the trailing root '.' optional.

nfeske, 2017-04-21:
I agree that inconsistency is bad.

But I'd go in the opposite direction by dictating that any path processed by VFS plugins must have a canonical form (i.e., leading slash). If this is the contract, the VFS plugins don't need to take defensive precautions as devised by your patch. This is arguably better because there will be many VFS plugins - possibly written by 3rd-parties - which may potentially miss sanitizing the path arguments. Tightening the interface relieves this code from the burden of sanitizing arguments.

Speaking of a "contract", the best approach would be to change the method interfaces by passing an `Absolute_path const &` as argument, instead of a `char const *`. The only downside of this approach is the fixed path limit of the `Absolute_path`. But I would ignore this problem as long as we are able to define a reasonable limit. Given that we don't have a system-global and hugely nested VFS, I can't think of encountering a situation where the current limit of 512 would stand in the way.

nfeske, 2019-11-19:
I'm closing this issue since it has become stale.

closed 2019-11-19 by nfeske