genodelabs/genode #2397: File_system: replace per-handle signals with notification packets

ehmry, 2017-04-17:
As incremental step towards VFS notifications (#1934), I would like to replace the per-handle signal context with notification packets at the File_system session interface. The will reuse the required packet handling signals to asynchronously notify clients and requires no additional signals during the lifetime of the session.

This is good for asynchronous VFS notifications because it reduces the per-handle state overhead, notifications packets only need to be correlated to VFS handles, which is already the case for I/O packets.

ehmry, 2017-04-20:
I have ``sigh`` removed from ``File_system::Session`` in favor of ``CONTENT_CHANGED`` at 5234a6be46ac42e61ee61969840a67b919194eac. There is a test of ``fs_rom`` signaling at ``run/fs_rom_update``.

chelmuth, 2017-05-02:
This looks great. I merged 6651fa4bb9e91f90f3e4fa25b111bb8966dd6fc1 and cfb78c19e33a40c2eb06ccfc85304264a5132ef5 to staging.

chelmuth, 2017-05-02:
@ehmry please try to fix all servers that depend on `sigh()`. I stumbled upon compiler error in *trace_fs* a second ago.

ehmry, 2017-05-02:
I have applied the changes to ``ram_fs`` to the other servers at 379d5750e26e672857704c6a784b4139f0179ce6. The notification framework from ``ram_fs`` is present in these servers, but I don't believe it is in use. I say leave them as they are and perhaps convert them to VFS plugins later.

nfeske, 2017-05-03:
Just for the record, I only used the notifications of ram_fs combined with fs_rom, and never tried using fs_rom with any other file-system server.

chelmuth, 2017-05-03:
@ehmry thanks for the fixup I merged just now.

chelmuth, 2017-05-03:
We still have a warning in the VFS server
```make
.../repos/os/src/server/vfs/main.cc: In member function 'void Vfs_server::Session_component::_process_packet_op(File_system::Packet_descriptor&)':
.../repos/os/src/server/vfs/main.cc:139:11: warning: enumeration value 'CONTENT_CHANGED' not handled in switch [-Wswitch]
    switch (packet.operation()) {
           ^
```

ehmry, 2017-05-03:
I left that as a reminder, but better to cleanly discard the packets, fixed at 3c022031cc3248f226f45f5477cb6b76d733b740.

ehmry, 2017-05-18:
@chelmuth I have a fixup for ``fs_rom`` at c6502b17e62358954d44da0439ba9fb151ec97f4.

The problem with staging is that the session dataspaces are re-allocated each time the client calls ``dataspace`` and the previous dataspace is invalidated. Dataspace is being called more than once on component binaries and this invalidates the mappings, triggering faults.

nfeske, 2017-05-18:
@ehmry I am afraid that your fixup above merely masks a problem. The real issue is the client-side misuse of the ROM session interface. Please have a look at the documentation of `Rom_session::dataspace`:

```
/**
 * Request dataspace containing the ROM session data
 *
 * \return  capability to ROM dataspace
 *
 * The capability may be invalid.
 *
 * Consecutive calls of this method are not guaranteed to return the
 * same dataspace as dynamic ROM sessions may update the ROM data
 * during the lifetime of the session. When calling the method, the
 * server may destroy the old dataspace and replace it with a new one
 * containing the updated data. Hence, prior calling this method, the
 * client should make sure to detach the previously requested dataspace
 * from its local address space.
 */
virtual Rom_dataspace_capability dataspace() = 0;
```

Tweaking the behavior of rom_fs to comply with wrong assumptions of some clients is the wrong direction. Let us better fix the clients. Can you point me to the components where you encountered the problem?

ehmry, 2017-05-18:
@nfeske There are a number of them but ``nitpicker`` triggers it reliably. It will track it down though, it took me a while to find the reason for these faults anyway.

ehmry, 2017-05-18:
@nfeske I have a test that triggers this on my ``fs_rom_dataspace`` branch at ``run/fs_rom_dataspace``. Does not happen with with linux.

ehmry, 2017-05-18:
c6502b17e62358954d44da0439ba9fb151ec97f4 does reduce allocation and mapping so I think it would still be a worthwhile improvement.

nfeske, 2017-05-23:
You are right that the patch is an improvement. For now, I'd still prefer to keep the original version as it uncovers issues like the one you found in ldso. With your optimization in place, I am afraid that such problems may still occur but much more sporadically and thereby harder to reproduce. Would you be fine with keeping the patch unmerged for a while?

ehmry, 2017-05-23:
@nfeske sure, I'm content that with the ldso performance boost for now.

closed 2017-05-31 by chelmuth

BorisMulder-CSL, 2017-09-22:
@nfeske 
> I only used the notifications of ram_fs combined with fs_rom, and never tried using fs_rom with any other file-system server.
Are there any plans on making this work with other servers as well, especially rump_fs? On long-term systems, the ability to store e.g. config files on persistent locations and being able to notify updates to clients through fs_rom could be a valuable addition.

ehmry, 2017-09-22:
@BorisMulder-CSL yes, there are plans, but I was hoping to do it as a feature of the VFS library and the rump plugin. See the end of #1934.

BorisMulder-CSL, 2017-09-25:
For us, that would be a really nice feature.
More so, it would even be enough if the client that writes to the file could just call a function to manually issue an update for all listening clients (such as fs_rom).

ehmry, 2017-09-25:
@BorisMulder-CSL my plan was to send updates on ``sync``, I suppose a empty append write or a truncate to the current size would do it.

BorisMulder-CSL, 2018-01-05:
@ehmry I have seen all kinds of activity on vfs, such as in #1934 . Does this feature work now, and if so, is the idea now to favor the rump_fs vfs plugin over the rump_fs server in all use cases?

Currently we have workarounds where we keep a copy in a ram_fs in order to support dynamic updates (as to not having to restart the system if a user changes, say, the audio volume). We would be glad to get rid of this :)

commit cfb78c19 (2017-05-02): File_system: replace per-handle signals with notification packets

Replace registration and signaling of per-handle signal capabilities
with CONTENT_CHANGED notification packets.

Fix #2397
commit e227c21c (2017-05-31): File_system: replace per-handle signals with notification packets

Replace registration and signaling of per-handle signal capabilities
with CONTENT_CHANGED notification packets.

Fix #2397