genodelabs/genode #2398: Capability quota accounting

nfeske, 2017-04-19:
Genode employs a resource-trading scheme for memory management. Under this regime, parent components explicitly assign memory to child components, and client components are able to "lend" memory to servers. (the details are described in the "Genode Foundations" book)

Even though capabilities are data structures (residing in the kernel), their costs cannot be accounted via Genode's regular memory-trading scheme because those data structures are - generally speaking - not easily extensible by the user land on top of the kernel. E.g., on Linux where we use file descriptors to represent capabilities, we are bound by the fd-limit of the kernel. On base-hw, the maximum number of capabilities is fixed at the build time of the kernel and used to dimension statically allocated data structures. Even on seL4 (which in principle allows user memory to be turned into kernel memory), the maximum number of capabilities is somehow limited by the ID namespace within core. For this reason, capabilities should be regarded as a limited physical resource from the component's point of view, very similar to how physical memory is modeled at a limited physical resource.

In the current version of Genode, any regular component triggers the allocation of capabilities whenever a RPC object or a signal context is created (via the `Pd_session::alloc_rpc_cap`, or `Pd_session::alloc_context`). As there is no limit of how many capabilities a component can allocate, a misbehaving component may exhaust the system-global capability space and thereby poses a denial-of-service threat.

The solution is the application of Genode's resource trading scheme to capabilities in addition to memory. It entails the following construction sites:

* Enhance PD session (`transfor_quota` analogously to `Ram_session`)
* Core implementation of `transfer_quota`
* Dimension initial cap quota of core/init depending on the platform
* Cap quota handling in base/child.cc
* Equip connections with `cap_quota` session args
* Handle cap_quota at the server side
  * Nitpicker
  * Core: ram, cpu, rm
  * Review the existing interfaces with respect to capability allocations (e.g., account for the dataspace capability needed to establish a packet stream)
* Out_of_caps exception
  * Session creation (parent interface, session state)
  * Session interfaces
* Transparent cap-quota upgrading via the `retry` pattern
* Init: assignment of cap quotas to start nodes, dynamic adjustments

m-stein, 2017-05-19:
Added commits:
6f45f57 fixup "Assign cap quotas in run scripts and recipes" (several scripts)
d812231 noux_net_lighttpd.run: heavily outdated, remove
309c7ad run + noux_gdb: fix source archiving

closed 2017-05-31 by chelmuth

m-stein, 2017-06-14:
@chelmuth Added a fixup:
1150818 nic_router test: raise timer cap quota