genodelabs/genode #2400: os/timer: interpolate time via timestamps

m-stein, 2017-04-26:


alex-ab, 2017-04-27:
> This commit adds the new Kernel::time syscall to base-hw. The syscall is
solely used by the Genode::Timer on base-hw as substitute for the
timestamp. This is because on ARM, the timestamp function uses the ARM
performance counter that stops counting when the WFI (wait for
interrupt) instruction is active.

Nice. What me actually bothers, is how is the master plan about the other ARM kernels in general ?
I mean we still/will have foc and (presumably) seL4 on ARM.  Are those counters ever programmable by userland without kernel help ? If not, I would expect extending all kernels by extra syscalls is not what we want in general (especially if we tend to use a verified seL4 kernel).

m-stein, 2017-04-27:
I forgot to mention what happens if we run on ARM with another Kernel than HW. In this case, the timestamp speed may continuously vary from almost 0 up to CPU speed. The Timer, however, only uses interpolation if the timestamp speed remained stable (12.5% tolerance) for at least 3 observation periods. Currently, one period is 100ms, so its 300ms. As long as this is not the case, Timer_session::elapsed_ms is called instead.

Anyway, it might happen that the CPU load was stable for some time so interpolation becomes active and now the timestamp speed drops. In the worst case, we would now have 100ms of slowed down time. The bad thing about it would be, that this also affects the timeout of the period. Thus, it might "freeze" the local time for more than 100ms.

On the other hand, if the timestamp speed suddenly raises after some stable time, interpolated time can get too fast. This would shorten the period but nonetheless may result in drifting away into the far future. Now we would have the problem that we can't deliver the real time anymore until it has caught up because the output of Timer::curr_time shall be monotone. So, effectively local time might "freeze" again for more than 100ms.

I think, it would be a solution to not use the Trace::timestamp on ARM w/o HW but a function whose return value causes the Timer to never use interpolation because of its stability policy.

m-stein, 2017-05-08:
@chelmuth I've updated the commit according to your suggestions.

m-stein, 2017-05-14:
@chelmuth Updated according to your comments
ef934bd os/timer: interpolate time via timestamps

ehmry, 2017-05-14:
Fix to get libc to compile at 26792e6551e30df6eadb3217041996849b7ca898

m-stein, 2017-05-18:
@chelmuth build fixups
7210f74 fixup "os/timer: interpolate time via timestamps" (make all)
bdbe5ac fixup "os/timer: interpolate time via timestamps" (input_filter)
2aa9452 fixup "os/timer: interpolate time via timestamps" (terminal)
870ac71 fixup "os/timer: interpolate time via timestamps" (nic_dump)
0134863 fixup "os/timer: interpolate time via timestamps" (nic_router)

m-stein, 2017-05-18:
@chelmuth one more for the build
991556c fixup "os/timer: interpolate time via timestamps" (make all #2)

chelmuth, 2017-05-19:
With current staging I get a whole bunch of log messages like follows when using Qemu.
```
[component] Warning: reading remote time is too slow
```
Could this be alleviated or even removed?

m-stein, 2017-05-19:
I changed the behavior to "if the latency is always bad, take the best latency you can get" and removed the warning:
56c1707 fixup "os/timer: interpolate time via timestamps" (qemu warnings)

m-stein, 2017-05-22:
@chelmuth another fixup:
c5b7159 fixup "os/timer: interpolate time via timestamps" (qemu warnings #2)

m-stein, 2017-05-23:
@chelmuth @nfeske These commits merge the Genode::Timer into the Timer::Connection, and set the direction to use only the interface of Periodic_timeout and One_shot_timeout for timing in the future:
```
ea47df1 timeout: document what to use and what not
a3251f0 timeout: move os/timeout.h to timer/timeout.h
8c131ab timeout: *_timeout to timer_session/connection.h
8d85685 timeout: merge os/time_source.h into os/timeout.h
e5a7347 timeout: rename timer*.cc timer_connection*.cc
806adeb timeout: Timer_time_source to Timer::Connection
da1a1b3 timeout.run: do not build test/util_mmio
cbc51f8 timeout: decrease TS_TO_US_RATIO_SHIFT to 4
c6794e4 timeout: Timer::Connection replaces Genode::Timer
0e56985 timeout: move lib dependency from LDSO to base lib
```

chelmuth, 2017-05-24:
@m-stein we still have issues with the interpolation commit on all 32-bit hardware platforms incl. linux. When I run `make -C x86_32 run/input_filter KERNEL=linux` I get many messages like
```
[init -> input_filter] Warning: Timestamp-to-time ratio too big MAX_TS=72057594037927935 ts=19843544827350 us=677190 result=7501509880 result_ul=3206542584
```
I instrumented *timer.cc* with
```diff
+++ b/repos/os/src/lib/timeout/timer.cc
@@ -43,6 +43,9 @@ void Timer_time_source::_update_interpolation_quality(unsigned long min_factor,
 unsigned long Timer_time_source::_ts_to_us_ratio(Timestamp     ts,
                                                  unsigned long us)
 {
+       Timestamp     const save_ts = ts;
+       unsigned long const save_us = us;
+
        /*
         * If the timestamp difference is to big to do the following
         * factor calculation without an overflow, scale both timestamp
@@ -67,7 +70,9 @@ unsigned long Timer_time_source::_ts_to_us_ratio(Timestamp     ts,
        if (result == result_ul) {
                return result_ul; }
 
-       warning("Timestamp-to-time ratio too big");
+       warning("Timestamp-to-time ratio too big MAX_TS=", (unsigned long long)MAX_TS,
+               " ts=", save_ts, " us=", save_us,
+               " result=", (unsigned long long)result, " result_ul=", result_ul);
        return ~0UL;
 }
```
Also, the input_filter test only succeeds in Qemu currently - all hardware platforms are broken.

chelmuth, 2017-05-24:
I rebased your commits on current staging and commented in my branch https://github.com/chelmuth/genode/tree/issue2400_timestamps

m-stein, 2017-05-29:
@chelmuth Added a bug fix for input filter and a test update:
9f20808 timeout test: test self re-scheduling one-shots
8a2af5c timeout: fix bug with self re-scheduling one-shots

chelmuth, 2017-05-29:
Okay, I put 881d2461877ff40613f600a273b9bd9e83311915 in the spirit of you bugfix on top of staging.

m-stein, 2017-05-30:
@chelmuth The Return Of The Fixups
a69b282 fixup "os/timer: interpolate time via timestamps" (calibrate earlier in test)
b7736e0 fixup "os/timer: interpolate time via timestamps" (sort lib/symbols/ld)
5a2eb89 fixup "os/timer: interpolate time via timestamps" (scheduler mode to connection)
437a61c fixup "os: removal of deprecated os/config.h (fix #2431)"
92c33d9 fixup "os/timer: interpolate time via timestamps" (raise tick length to 500 ms)
4f2545a fixup "os/timer: interpolate time via timestamps" (no Microseconds::max in libc task)
6f2dd21 fixup "os/timer: interpolate time via timestamps" (enable scheduler in timer driver)
1870e84 fixup "os/timer: interpolate time via timestamps" (do timer tick w/o interpolation)
0472436 fixup "os/timer: interpolate time via timestamps" (timeouts to namespace Timer)
aa3f9a0 fixup "os/timer: interpolate time via timestamps" (test self re-scheduling one-shots)
b5ddbc2 fixup "os/timer: interpolate time via timestamps" (documentation)
9f09ae0 fixup "os/timer: interpolate time via timestamps" (os/timeout.h to timer/timeout.h)
567ce98 fixup "os/timer: interpolate time via timestamps" (timeouts to timer_session/connection.h)
b592f0f fixup "os/timer: interpolate time via timestamps" (get rid of os/time_source.h)
bc1a24f fixup "os/timer: interpolate time via timestamps" (rename timer*.cc)
eb31d79 fixup "os/timer: interpolate time via timestamps" (time source to Timer::Connection)
e7d49fb fixup "os/timer: interpolate time via timestamps" (no util_mmio in run script)
afc923e fixup "os/timer: interpolate time via timestamps" (decrease TS_TO_US_RATIO_SHIFT)
ec79dcf fixup "os/timer: interpolate time via timestamps" (Timer to Timer::Connection)
e6eb388 fixup "os/timer: interpolate time via timestamps" (move to base lib)

m-stein, 2017-06-15:
@chelmuth Two more fixes:
2e5f359 timeout test: improve documentation
5440255 timeout test: consider time shift between sessions

m-stein, 2017-06-20:
@chelmuth These commits solve problems with panda/x86_32 and are tested on 51ab804 by mogli:
5e5736e hw cortex_a9: fix bug in kernel timer
c51592b timeout lib: dynamic interpolation-factor shift
96ebe87 timeout test: result buffer as struct

nfeske, 2017-06-21:
Unfortunately, 94095a27acd33368a60207a3b3de688af78fe6d7 seems to introduce an unwelcome boot delay. In my setup (Turmvilla), the system seems to "hang" right after NOVAs initial messages but finally continues. Sometimes, it gets stuck at this point and prints a message about the TSC calibration from time to time but does not proceed.

Is there a way to avoid the delay? In my opinion - compared with the prior version - the change spoils the experience of working/testing with NOVA quite a bit.

alex-ab, 2017-06-21:
> Unfortunately, 94095a2 seems to introduce an unwelcome boot delay. In my setup (Turmvilla), the system seems to "hang" right after NOVAs initial messages but finally continues. Sometimes, it gets stuck at this point and prints a message about the TSC calibration from time to time but does not proceed.

Just to note, the described behavior triggers on a hard setup for timing -  Genode@Nova in Qemu in Virtualbox on Genode@Nova. Timing wise - anything can happen.

Nevertheless, try https://github.com/alex-ab/genode/commit/36190ac0af4303ed02550aa33282d3c8eac0b34e. Does it help ? @m-stein: does this commit have negative impacts to the timeout test  (in qemu and x201) ?

m-stein, 2017-06-23:
I re-opened the issue because the timeout test has brought up some timing issues on different platforms that are yet to fix.

m-stein, 2017-06-23:
On platforms that use the PIT timer driver, 'Timer::Connection::elapsed_ms' is pretty inprecise/unsteady (up to 3 ms deviation) for a reason that is not clearly determined yet. On Fiasco and Fiasco.OC, that use kernel timing, it is the same. So, on these platforms, our locally interpolated time seems to be fine but the reference time is bad.

I measured the execution time for a fixed for loop in different constellations via the TSC and via 'elapsed_ms' (the TSC frequency was invariant and known). These are the results:

```
hw, 64 bit, in kernel (should be the most precise)
1: timestamp 262500 us
2: timestamp 262500 us
3: timestamp 262500 us
4: timestamp 262500 us
5: timestamp 262500 us
6: timestamp 262500 us
7: timestamp 262500 us
8: timestamp 262500 us
9: timestamp 262500 us

nova, 32 bit, in core
1: timestamp 262520 us
2: timestamp 262520 us
3: timestamp 262519 us
4: timestamp 262528 us
5: timestamp 262520 us
6: timestamp 262519 us
7: timestamp 262523 us
8: timestamp 262520 us
9: timestamp 262519 us

nova, 64 bit, in core
1: timestamp 262521 us
2: timestamp 262519 us
3: timestamp 262519 us
4: timestamp 262520 us
5: timestamp 262520 us
6: timestamp 262521 us
7: timestamp 262522 us
8: timestamp 262521 us
9: timestamp 262519 us

nova, 32 bit, in app
[init -> app] 1: timestamp 262526 us, timer 263000 us
[init -> app] 2: timestamp 262520 us, timer 263000 us
[init -> app] 3: timestamp 262521 us, timer 263000 us
[init -> app] 4: timestamp 262547 us, timer 263000 us
[init -> app] 5: timestamp 262520 us, timer 263000 us
[init -> app] 6: timestamp 262521 us, timer 263000 us
[init -> app] 7: timestamp 262520 us, timer 263000 us
[init -> app] 8: timestamp 262535 us, timer 263000 us
[init -> app] 9: timestamp 262525 us, timer 263000 us

okl4, 32 bit, in app
[init -> app] 1: timestamp 263916 us, timer 264000 us
[init -> app] 2: timestamp 263955 us, timer 265000 us
[init -> app] 3: timestamp 263953 us, timer 264000 us
[init -> app] 4: timestamp 263939 us, timer 265000 us
[init -> app] 5: timestamp 263947 us, timer 264000 us
[init -> app] 6: timestamp 263917 us, timer 265000 us
[init -> app] 7: timestamp 263959 us, timer 265000 us
[init -> app] 8: timestamp 263970 us, timer 264000 us
[init -> app] 9: timestamp 263972 us, timer 265000 us

foc, 64 bit, in app
[init -> app] 1: timestamp 263494 us, timer 264000 us
[init -> app] 2: timestamp 263495 us, timer 264000 us
[init -> app] 3: timestamp 263492 us, timer 265000 us
[init -> app] 4: timestamp 263492 us, timer 264000 us
[init -> app] 5: timestamp 263523 us, timer 264000 us
[init -> app] 6: timestamp 263530 us, timer 264000 us
[init -> app] 7: timestamp 263524 us, timer 264000 us
[init -> app] 8: timestamp 263538 us, timer 265000 us
[init -> app] 9: timestamp 263532 us, timer 264000 us

foc, 32 bit, in app
[init -> app] 1: timestamp 263390 us, timer 268000 us
[init -> app] 2: timestamp 263384 us, timer 268000 us
[init -> app] 3: timestamp 263386 us, timer 268000 us
[init -> app] 4: timestamp 263387 us, timer 268000 us
[init -> app] 5: timestamp 263417 us, timer 268000 us
[init -> app] 6: timestamp 263469 us, timer 268000 us
[init -> app] 7: timestamp 263466 us, timer 268000 us
[init -> app] 8: timestamp 263461 us, timer 268000 us
[init -> app] 9: timestamp 263461 us, timer 268000 us

pistachio, 32 bit, in app
[init -> app] 1: timestamp 263953 us, timer 265000 us
[init -> app] 2: timestamp 264005 us, timer 265000 us
[init -> app] 3: timestamp 264020 us, timer 264000 us
[init -> app] 4: timestamp 264015 us, timer 265000 us
[init -> app] 5: timestamp 264046 us, timer 264000 us
[init -> app] 6: timestamp 263972 us, timer 265000 us
[init -> app] 7: timestamp 264045 us, timer 264000 us
[init -> app] 8: timestamp 264035 us, timer 265000 us
[init -> app] 9: timestamp 264048 us, timer 265000 us

fiasco, 32 bit, in app
[init -> app] 1: timestamp 263789 us, timer 264000 us
[init -> app] 2: timestamp 263789 us, timer 265000 us
[init -> app] 3: timestamp 263793 us, timer 264000 us
[init -> app] 4: timestamp 263798 us, timer 265000 us
[init -> app] 5: timestamp 263797 us, timer 264000 us
[init -> app] 6: timestamp 263827 us, timer 265000 us
[init -> app] 7: timestamp 263839 us, timer 264000 us
[init -> app] 8: timestamp 263834 us, timer 265000 us
[init -> app] 9: timestamp 263825 us, timer 265000 us
```

As the hw kernel measurement shows, the loop actually consumes about 262500 us. On all kernels except nova (where kernel timing is used), the timer results deviate remarkably from this desired value (up to 6 ms) and are unsteady ( > 1 ms).

If we, however, replace the implementation of 'elapsed_ms' on the "bad" platforms with an implementation that solely reads the TSC and divides it by the (in this case known) TSC frequency, 'elapsed_ms' becomes accurate and the timeout test succeeds reliably.

I have also measured the latencies in the PIT implementation of 'elapsed_ms' on okl4:

```
// RPC call inside app
Timer::Connection::elapsed_ms();  // =  22-27 us (no load) / 30-38 us (with load)

// implementation inside timer
Duration Timer::Time_source::curr_time()
{
	// ... the rest = 0 us
	uint16_t const curr_counter = _read_counter(&wrapped); // = 19-21 us (no load) / 18-34 us (with load)
	// ... the rest = 0 us
}
```

m-stein, 2017-06-23:
The 'Time_source::_read_counter' seems to be the most expensive part and this function does some RPCs on the IOPORT session to core. This small latency shouldn't be the problem. Anyway, I moved the IOPORT access from core to the timer driver to be safe. It didn't fix the problem. So I did the loop measurement of above again directly in the PIT timer driver with the '_read_value' method and without other activites interrupting (also no PIT 54ms wrap timeout itself). The results were again around 264-265 ms.

m-stein, 2017-06-23:
@chelmuth 3 minor fixes:
ce769dd timeout test: prioritize timer driver over test
0e2e0aa timer okl4: remove deprecated const_cast
dd2b4e0 pit/fiasco timeout: raise time error tolerance

alex-ab, 2017-06-23:
> The 'Time_source::_read_counter' seems to be the most expensive part and this function does some RPCs on the IOPORT session to core. This small latency shouldn't be the problem. Anyway, I moved the IOPORT access from core to the timer driver to be safe.

4 inb/outb calls respectively Genode RPCs are at least full 8 address space switches [inclusively flushing TLBs and caches (no tagged TLB without PCID which got added late and solely in 64bit)]. Depending on the kernel, you may even in core don't do the assembler inb/outb instruction but instead make a kernel syscall. And finally the scheduler may play bad with you.

I would call it really expensive and worse regarding latency and timing.

m-stein, 2017-08-08:
I added a commit series that fixes some bugs and removes the hardware-timer work-around from the RPI USB driver:
7487622 usb rpi: get rid of local hardware timer
f6f577a timer connection: fast initial calibration
c830582 timer connection: relax factor shifting
ca47d6b hw rpi: fix bug that caused bad timer precision
d990fcc timer connection: always work with microseconds
ebe4a82 timer session: add elapsed_us method

chelmuth, 2017-08-23:
In https://github.com/genodelabs/genode/blob/staging/repos/os/src/lib/timeout/timer_connection_time.cc#L75 `_us == us` is true in our netperf_lwip_wifi tests. This breaks *wifi_drv* with a division by zero at https://github.com/genodelabs/genode/blob/staging/repos/os/src/lib/timeout/timer_connection_time.cc#L151. Any ideas?

chelmuth, 2017-08-23:
BTW, with the following patch the test succeeds.
```diff
+++ b/repos/os/src/lib/timeout/timer_connection_time.cc
@@ -72,6 +72,10 @@ void Timer::Connection::_update_real_time()
        }
 
        /* determine timestamp and time difference */
+if (_us == us) {
+       warning("_us == us (", _us, ")");
+       return;
+}
        unsigned long const us_diff = us - _us;
        Timestamp           ts_diff = ts - _ts;
 
```

m-stein, 2017-08-23:
@chelmuth Sorry for my sparse state feedback - I started debugging timeout problems this morning following our offline discussion yesterday. The division by zero is already fixed and I'm currently at the problem behind insufficient time precision on hw/x86_64:

1399c29 test/timeout: use elapsed_us instead of elapsed_ms
d69c847 timer connection: fix division by null

m-stein, 2017-08-23:
@chelmuth I could solve problems with the PIT timer in general which caused the latest test fails on x86. I was also able to remove the exception that we made for PIT platforms in the timeout test regarding average-error limits:

703bd8f timeout test: remove error-limit exception for PIT
35ad581 timer: generic timer_ticks_to_us implementation
e530af5 timer pit: fix precision reduction to milliseconds
4e53d46 hw pit: fix precision reduction to milliseconds
629d4fb test/timeout: use elapsed_us instead of elapsed_ms
c99add4 timer connection: fix division by null

m-stein, 2017-08-24:
@chelmuth @cnuke Two more commits. This solves the problem with the assertion about 'ticks_per_ms':

920ea99 timer/util: generic TIMER_MIN_TICKS_PER_MS
3276a77 hw lapic: find best frequency dynamically

cnuke, 2017-08-24:
@m-stein thanks, fixes my issue :+1:

m-stein, 2017-08-24:
Oops, missed the relics of antiquity:

2bf56f1 fixup "timer: generic timer_ticks_to_us implementation"

m-stein, 2017-08-26:
This should quickfix our problems with older x86 machine:
 
1165256 x86 timeout test: consider instable tsc (quickfix)

m-stein, 2017-08-30:
@chelmuth These should fix the problems with the PIT on x86-32 on Qemu:
eca50b4 timer pit: handle and display bad latency
8e909f3 nic_router & timer* tests: prioritize timer driver

m-stein, 2017-09-05:
@chelmuth Another fix for the PIT, thanks to a hint by @alex-ab:
d3e463d timer pit: fix bug in multi-wrap detection

chelmuth, 2017-09-07:
Merged d6051bc and 3d1d671.

chelmuth, 2017-11-06:
The work is integrated into our master branch. Open further issues for specific fixes/bugs.

closed 2017-11-06 by chelmuth