genodelabs/genode #2403: Limit SPEC identifiers to specify ISA

skalk, 2017-04-28:
Currently, there exists a zoo of 'SPEC' identifiers to specify devices, SoCs, boards, architectures, or features, like the usage of TrustZone technology. Those 'SPEC' identifiers are either used to enforce or deny compilation of components, see the 'REQUIRE' mechanism, or to implicitly add additional include paths. 

Both use cases can be scrutinized partly. The implicit addition of 'SPEC'-ific include paths, in which homonymous headers reside, might simplify the writing of generic code snippets. But when reading the code it hinders to find the corresponding headers with an increasing number of 'SPEC'-ifiers and dimensions. As a consequence this mechanism should be used with care.

Second, the 'REQUIRE' mechanism is used not only to compile for the right ISA only, but also denotes the context in which the component will be used implicitly. For instance, there are drivers that compile/link to target the ARM architecture in general, but require a specific board 'SPEC' identifier to be built actually. This was due to the fact that drivers are named identically, and otherwise might override each other. Nowadays as Genode components can be build platform-agnostic, we like to limit the build directories to the actual ISA. By renaming drivers and components accordingly, we can get rid of a lot of current 'SPEC' identifiers. And by defining components and inclusions more explicitly, Genode's code base hopefully gets more easy to read by outsiders and developers.

skalk, 2017-04-28:
@nfeske @chelmuth would you mind to review commit 392303a for staging?

skalk, 2017-05-02:
@nfeske @chelmuth the next two commit moving in the described direction d3104a6 and 46e9f12

chelmuth, 2017-05-02:
I merged 392303a5cd6597fb3b3f34b7362c9bd64dbb7a16 as e1ad7454845cf919e9406b96f9e9f2d992fe7bf3 for a first step.

skalk, 2017-05-03:
@chelmuth thanks, the vmm for Arndale striked tonight, here is a fixup commit d3da91a7c1d9a8d5a6fd537ed15fd64664d53655 for it.

skalk, 2019-04-15:
I think it is good to take this issue into your mindset when working on issue like #2190 and #3180, or when introducing new platforms and drivers. But there will be no specific commit "solving" this issue. Therefore I close it for now.

closed 2019-04-15 by skalk