genodelabs/genode #2405: Fiasco.OC: update to recent revision

skalk, 2017-05-03:
Instead of debugging quite old code to run with the new toolchain (see issues #2372 and #2390) it would be far better to use the recent version of Fiasco.OC, sigma0 and bootstrap that already solved compiler issues and certain problems we noticed in the past.

skalk, 2017-05-03:
To ease the maintenance of Fiasco.OC, it would be good to leave away certain of its patches. For instance we add support for i.MX53, which is quite good covered by our own kernel. On the other hand the GDB support is not fully complete, but quite invasive. Last but not least we add support to use the kernel debugger out of a Genode Terminal. I would vote for removing these items at this time, which highly facilitates updates in the future. Moreover, those features are still in the history, and can be re-activated if someone really wants them.

nfeske, 2017-05-03:
@skalk, I agree with all of your points.

Besides removing the kernel changes, I would very much appreciate you reviewing the Genode source source tree regarding the stale features (run scripts, documentation snippets).

skalk, 2017-05-04:
@nfeske sure!

skalk, 2017-05-11:
@chelmuth commits d19ebac and 8294723 solve the issue. Please commit those to the buildbot ;-)

chelmuth, 2017-05-18:
I discovered that we have to remove foc_imx53 from our nightly tests and during this had a look at *tool/create_builddir*. Am I right, that it still references foc for imx53 in line 142? @skalk could you please clarify?

skalk, 2017-05-29:
@chelmuth above commit  ba84dbb makes Fiasco.OC/L4Re less noisy

commit ccf34f6a (2017-05-12): foc: update to recent revision r72 (fix #2405)
commit 48877d90 (2017-05-30): fixup "foc: update to recent revision r72 (fix #2405)"
closed 2017-05-31 by chelmuth