genodelabs/genode #2406: migrate netperf tests to the socket_fs

ehmry, 2017-05-03:
At present there are two Netperf test for Lxip and LwIP, each linked to the respective libc plugin. Now that the socket_fs is maturing to replace libc plugins, it is time for these tests to use VFS plugins.

ehmry, 2017-05-04:
f714177b5f2315e30a16a35143623b1fb6611b92 and 58cf64f529fc3be975d641624aaf236885c34c7a makes the Lxip plugin work with Netperf.

LwIP and Netperf is still in progress, there were some changes to libc and VFS since I last worked on it.

ehmry, 2017-05-12:
Rebasing...

ehmry, 2017-05-12:
Closes on my ``issue2335`` (#2335) branch or with 1719cdae8050624ad62125656aaf469b56ce8021, 7e02583c91684eda6db3e16088a94a0478f5fe4f, and b3158b209f22aefb5ac770aaeba3aae00f15da19.

chelmuth, 2017-10-19:
Is this implicitly solved by #2535?

ehmry, 2017-10-19:
Yes, it is.

chelmuth, 2017-10-20:
CLosing the issue cause it is now addressed by #2535.

closed 2017-10-20 by chelmuth