genodelabs/genode #2410: VFAT VFS plugin

ehmry, 2017-05-10:
Relating to UEFI support #2242, I converting the ``ffat_fs`` to a VFS plugin last year before the hack'n'hike but never finished the job.

chelmuth, 2017-05-22:
It seems the topic branch of this issue has gone. @ehmry would you mind to recreate it or is it obsolete?

ehmry, 2017-05-22:
Yes, I've updated my ``issue2410`` branch. I've been using this plugin without problems on my T420 but I haven't written enough to be confident that it is as robust as rump_fs.

ehmry, 2017-05-22:
Now that the VFS can be reconfigured there ought to be an attribute that can be set to unmount these sub-file-systems.

ehmry, 2017-05-24:
I have a patch to implement the necessary periodic sync using @m-stein's new timeout facility at 485e54892e0e73383fc150033ed80fc2c41ebd85, and a patch to remove ``libc_ffat`` at 91e0b5fa870c30fe8bc41025eba15c824cb7731d.

The libc documentation could use an update before the next release, see repos/libports/doc/libc.txt.

chelmuth, 2017-07-05:
Today, I gave your branch a spin and after a second look at the code I have some points to discuss.

I stumbled over the `Rtc_connection` in the FatFs-library backend which is the wrong direction in the long run for two reasons. First, the RTC is a pretty slow device which is should not be queried repeatedly at runtime and, second, there's no way to configure the notion of real time for the library beyond limiting RTC access for the whole component. It would be much better to provide means to configure a VFS path to a kind of *timeofday* file in the VFS plugin which can be read on `get_fattime()`. But, this requires the VFS to provide the root directory to its plugins on instantiation to permit access to VFS nodes. The *timeofday* file could then query RTC initially and afterwards use a timer connection for offsetting.

Also, I've got the feeling we should use consistent naming for the library/plugin, either ffat or fatfs. From the authors website I tend to change the current naming to fatfs.

ehmry, 2017-07-05:
I am fine with dropping the timestamp support, I hadn't considered the performance implications and the timestamps aren't available from the VFS anyway. As for the name I prefer fatfs as well.

chelmuth, 2017-07-25:
@ehmry I'm in the merging process for this plugin and also have some not yet public commits regarding renaming and restructuring. I expect the merge to finish within the next two days and contain your recent updates too.

ehmry, 2017-07-25:
Ok, my issue2410 branch is consistent with all my changes.

To add my own comment, the ``dirent`` performance is quite poor and it shows when using Noux. I would consider adding a cache for directory entries, but it can wait.

nfeske, 2017-07-25:
The main reason for poor dirent performance may be that each directory entry involves a full round trip between the noux client and server (see the implementation of `getdirentries` in ports/src/lib/libc_noux/plugin.cc).

Extending the Noux `dirent` syscall to query multiple entries as a batch would be a good start to improve the performance. The next stop would be the VFS library, which has the same limitation (the interface of `Directory_service::dirent` queries only a single element, too).

ehmry, 2017-07-25:
Yes, reading multiple dirents would be nice, and a good reason to start benchmarking the VFS more closely.

FatFS dirent performance is especially bad because directories cannot be "seeked", so ``num_dirent`` is a full traversal of dirents, and for each ``dirent`` all preceding entries are traversed. And I don't think there is directory caching over the raw blocks.

nfeske, 2017-07-25:
Now I understand your remark about the need for caching. I fully agree.

ehmry, 2017-07-31:
@chelmuth My branch ``issue2410_fatfs`` incorporates all my recent changes, has been tested on real hardware, and I believe it is ready for staging.

One point that eventually needs correction is that the FatFS library does not support read-only mounting. This is fine, so long as the ``Block`` component stack supports read-only policies, which I will check.

ehmry, 2017-08-11:
I have not tested ``fatfs_fs`` but I've found that the ``fatfs`` vfs plugin corrupts files larger than a several megabytes on when used on real hardware. I don't doubt that it reads files consistently, and because the library was never intended for high performance, disabling writes before the next release might be practical. My experience with the plugin as a backend for a desktop system has not been good, so I think time is better spent on other file-system projects, and if there is a need for writing to FAT with a light-weight plugin, this problem could be investigated.

There is also FAT support in ``rump_fs``, but I don't have any experience with it.

ehmry, 2017-08-17:
I'm getting consistent writes now with fixup 99fc41227e9f96a08e206460a1ebeeca0777f1c4. I believe the problem is related to seeking a file with unsynced writes. Like I said, not a lot of caching going on here.

chelmuth, 2017-08-17:
What's your current rationale behind the lock guards? Isn't it the duty of the user to protect the VFS instance from concurrent access?

ehmry, 2017-08-17:
It is, but I worry that corruption can happen easily if a mistake were made.

ehmry, 2017-09-01:
Closed in 17.08 release.

closed 2017-09-01 by ehmry