genodelabs/genode #2411: hw + pbxa9 + noux_terminal_fs: undefined references

m-stein, 2017-05-14:
When making run/noux_terminal_fs for hw on pbxa9 with staging: 
```
including /home/lypo/genodelabs/genode/git/tool/run/power_on/qemu
including /home/lypo/genodelabs/genode/git/tool/run/log/qemu
including /home/lypo/genodelabs/genode/git/tool/run/image/iso
including /home/lypo/genodelabs/genode/git/tool/run/boot_dir/hw
including /home/lypo/genodelabs/genode/git/repos/ports/run/noux_terminal_fs.run
building targets: core init drivers/timer noux lib/libc_noux drivers/framebuffer drivers/input server/terminal server/ram_fs server/nitpicker app/pointer server/nit_fb noux-pkg/bash noux-pkg/coreutils 
spawn make core init drivers/timer noux lib/libc_noux drivers/framebuffer drivers/input server/terminal server/ram_fs server/nitpicker app/pointer server/nit_fb noux-pkg/bash noux-pkg/coreutils
make[1]: Entering directory '/home/lypo/genodelabs/genode/git/build/pbxa9'
checking library dependencies...
Skip target drivers/framebuffer/intel because it requires x86
Skip target drivers/framebuffer/spec/exynos5 because it requires exynos5
Skip target drivers/framebuffer/spec/imx53 because it requires imx53
Skip target drivers/framebuffer/spec/omap4 because it requires omap4
Skip target drivers/framebuffer/spec/rpi because it requires rpi
Skip target drivers/framebuffer/spec/sdl because it requires linux sdl
Skip target drivers/framebuffer/vesa because it requires x86
Skip target drivers/input/spec/imx53 because it requires imx53
Skip target drivers/input/spec/ps2/x86 because it requires x86
Skip target drivers/timer/fiasco because it requires fiasco
Skip target drivers/timer/foc because it requires foc
Skip target drivers/timer/linux because it requires linux
Skip target drivers/timer/nova because it requires nova
Skip target drivers/timer/pit because it requires x86
  Library syscall-hw
  Library ldso-startup
  Library alarm
  Library cxx
  Library libc-stdlib
  Library libc-locale
  Library libc-string
  Library libc-inet
  Library libc-stdio
  Library libc-gdtoa
  Library libc-stdtime
  Library libc-regex
  Library libc-setjmp
  Library libc-mem
  Library libc-compat
  Library timed_semaphore
  Library libc-isc
  Library libc-resolv
  Library libc-nameser
  Library blit
  Library libc-rpc
  Library libc-net
  Library libc-gen
  Library ld
  Library timeout
  Library base
  Library vfs
  Program drivers/framebuffer/spec/pl11x/pbxa9/fb_drv
  Library base-hw-common
  Program drivers/input/dummy/dummy_input_drv
  Program app/pointer/pointer
  Program drivers/input/spec/ps2/pbxa9/ps2_drv
  Program drivers/timer/hw/hw_timer_drv
  Program init/init
  Program server/nit_fb/nit_fb
  Program noux/minimal/noux
  Program server/nitpicker/nitpicker
  Program server/ram_fs/ram_fs
  Program server/terminal/terminal
  Library libc
  Library libc_noux
  Library posix
  Program lib/libc_noux/libc_noux
  Library lwip
  Library core-hw
    COMPILE  version.o
  Library libc_lwip
    MERGE    core-hw.lib.a
  Library libc_lwip_nic_dhcp
  Library libm
  Program noux/net/noux_net
    LINK     noux_net
  Program noux-pkg/bash/bash
  Program noux-pkg/coreutils/coreutils
    CONFIG   bash
    CONFIG   coreutils
  Program core/hw/core
/home/lypo/genodelabs/genode/git/build/pbxa9/var/libcache/libc_lwip/libc_lwip.lib.a(plugin.o):(.data.rel.ro+0x94): undefined reference to `Libc::Plugin::getdirentries(Libc::File_descriptor*, char*, unsigned int, long long*)'
/home/lypo/genodelabs/genode/git/build/pbxa9/var/libcache/libc_lwip/libc_lwip.lib.a(plugin.o):(.data.rel.ro+0xb4): undefined reference to `Libc::Plugin::mmap(void*, unsigned int, int, int, Libc::File_descriptor*, long long)'
/home/lypo/genodelabs/genode/git/build/pbxa9/var/libcache/libc_lwip_nic_dhcp/libc_lwip_nic_dhcp.lib.a(plugin.o):(.data.rel.ro+0x94): undefined reference to `Libc::Plugin::getdirentries(Libc::File_descriptor*, char*, unsigned int, long long*)'
/home/lypo/genodelabs/genode/git/build/pbxa9/var/libcache/libc_lwip_nic_dhcp/libc_lwip_nic_dhcp.lib.a(plugin.o):(.data.rel.ro+0xb4): undefined reference to `Libc::Plugin::mmap(void*, unsigned int, int, int, Libc::File_descriptor*, long long)'
collect2: error: ld returned 1 exit status
/home/lypo/genodelabs/genode/git/repos/base/mk/prg.mk:179: recipe for target 'noux_net' failed
make[3]: *** [noux_net] Error 1
var/libdeps:825: recipe for target 'noux_net.prg' failed
make[2]: *** [noux_net.prg] Error 2
make[2]: *** Waiting for unfinished jobs....
^CMakefile:309: recipe for target 'run/noux_terminal_fs' failed
make: *** [run/noux_terminal_fs] Error 130
```

chelmuth, 2017-05-15:
Isn't this fixed by c0fcaf00b3252c9e758c75fe3173eccc2950aed9?

m-stein, 2017-05-16:
I've tried it directly on the commit and on staging. It's still the same.

chelmuth, 2017-05-18:
Was the issue fixed by the tool-chain update and can be closed?

m-stein, 2017-05-19:
I stumbled over pretty much the same problem when building the LXIP UDP test client for arm and x86_32 with current staging (x86_64 is fine, kernel doesn't matter):

```
lypo@bird:~/genodelabs/genode/git/build/panda$ make test/lxip/udp_client
checking library dependencies...
  Library libc-stdlib
  Library libc-locale
  Library libc-string
  Library libc-stdio
  Library libc-inet
  Library libc-gdtoa
  Library libc-stdtime
  Library libc-regex
  Library libc-compat
  Library libc-setjmp
  Library libc-mem
  Library ldso-startup
  Library libc-isc
  Library lxip_include
  Library libc-gen
  Library libc-resolv
  Library ld
  Library libc-nameser
  Library libc-rpc
  Library base
  Library vfs
  Library libc-net
  Library libc
  Library libc_resolv
  Library lxip
  Library libc_lxip
  Program test/lxip/udp_client/test-lxip_udp_client
    LINK     test-lxip_udp_client
/home/lypo/genodelabs/genode/git/build/panda/var/libcache/libc_lxip/libc_lxip.lib.a(plugin.o):(.data.rel.ro+0x9c): undefined reference to `Libc::Plugin::getdirentries(Libc::File_descriptor*, char*, unsigned int, long long*)'
/home/lypo/genodelabs/genode/git/build/panda/var/libcache/libc_lxip/libc_lxip.lib.a(plugin.o):(.data.rel.ro+0xbc): undefined reference to `Libc::Plugin::mmap(void*, unsigned int, int, int, Libc::File_descriptor*, long long)'
/usr/local/genode-gcc-after-17-05/bin/../lib/gcc/arm-none-eabi/6.3.0/../../../../arm-none-eabi/bin/ld generated: undefined reference to `Libc::Plugin::getdirentries(Libc::File_descriptor*, char*, unsigned int, long long*)'
/usr/local/genode-gcc-after-17-05/bin/../lib/gcc/arm-none-eabi/6.3.0/../../../../arm-none-eabi/bin/ld generated: undefined reference to `Libc::Plugin::mmap(void*, unsigned int, int, int, Libc::File_descriptor*, long long)'
collect2: error: ld returned 1 exit status
/home/lypo/genodelabs/genode/git/repos/base/mk/prg.mk:179: recipe for target 'test-lxip_udp_client' failed
make[2]: *** [test-lxip_udp_client] Error 1
var/libdeps:462: recipe for target 'test-lxip_udp_client.prg' failed
make[1]: *** [test-lxip_udp_client.prg] Error 2
Makefile:272: recipe for target 'gen_deps_and_build_targets' failed
make: *** [gen_deps_and_build_targets] Error 2
```

m-stein, 2017-05-19:
This is already with the new tool chain.

chelmuth, 2017-05-19:
Sorry, but I can't confirm this issue. Rebuilt *test/lxip/udp_client* just now for staging/x86_32. Please consult @cproc to identify the problem on your host.

chelmuth, 2017-05-19:
I'm using the preliminary prebuilt *genode-gcc-17.05.4.tar.xz* from buildbot.

m-stein, 2017-05-19:
Sorry, my fault, had a problem with the ccache.

closed 2017-05-19 by m-stein