genodelabs/genode #2412: Planning of Hack'n'Hike 2018

m-stein, 2017-05-16:
To us, the Hack'n'Hike 2017 was a big success, like in the past years. We had a great hike through the Zittauer mountain landscape, enjoyed the view from atop of several extraordinary rock formations, and visited the beautiful monastery ruin in Oybin.

From the hacking perspective, probably the most significant project was the Quake 3 scenario of @cnuke and @ssumpf. They achieved to let us play Quake 3 on Genode in LAN multiplayer in the end. With fluent 70-120 FPS using the Intel GPU driver and low-latency sound. That was a great experience and a lot of fun. Furthermore, there were projects working on a Freeciv port, Raspberry Pi 3 support, native i.MX USB support, tooling for the new depot/package system, Virtualbox 5 support in Muen, and more ... .

That said, we'd like to invite you to continue hacking and hiking with us in 2018 ;-) This issue is meant as base for organizing, joining in, and getting informed. So, please feel free to post here if you have questions or suggestions regarding the Hack'n'Hike and, if you're interested, keep an eye on the issue as we will post all news in here.

chelmuth, 2017-07-05:
I'd love to move the next event back to Saxon Switzerland. Does anybody know the Zirkelstein Resort (http://www.zirkelsteinresort.de, http://www.openstreetmap.org/#map=16/50.8761/14.2173&layers=C)? It might be a candidate with those nice Bungalows and the Czech border just at a stone's throw.

nfeske, 2017-07-05:
I am fond of Forsthaus Bielatal: http://www.forsthaus-bielatal.de

nfeske, 2017-07-05:
Another very nice looking option: https://burg-schoena.de/galerie/

chelmuth, 2017-07-05:
Maybe http://www.felswelten.de/

m-stein, 2017-08-04:
@chelmuth I vaguely remember that we were in contact with the Zirkelstein Resort already some years ago and that for reasons it was not suitable for the Hack'n'Hike.

I have sent requests to the other three locations.

chelmuth, 2017-09-25:
Are there any opinions or even helpful hands, so we can finalize a date and location for the next event before December? I'm afraid it will become harder to find a nice accommodation that fits our needs the later we start the booking.

m-stein, 2017-09-25:
The Zirkelsteinresort is fully booked until 2019  o_O
As each of the other three options had its downs, I'll start looking out again for alternatives.

m-stein, 2017-11-30:
### Date and Accommodation

We finished reservation of the next Hack'n'Hike accommodation!

#### Location

Forsthaus Bielatal, Forsthof 1, 01824 Rosenthal-Bielatal, district Bielatal
- [Homepage](http://www.forsthaus-bielatal.de/_forsthaus-bielatal/index.html)
- [Open Street Maps](https://www.openstreetmap.org/?mlat=50.87345&mlon=14.04835#map=18/50.87345/14.04835)

#### Time
- Check-in as of 3 pm on May 25th
- Check-out till 11 am on May 28th

#### Room
- 7 bedrooms for 16 participants 
- A self-catering kitchen
- A lockable seminar room
- All exclusively used by us during the Hack'n'Hike

#### Misc
- Volume-limited Internet will be available
- Bedding and towels will be provided by the Forsthaus Bielatal
- You can park your car at the site

#### Catering
- Genode Labs will provide catering for breakfast and dinner in the Forsthaus Bielatal
- For lunch, we normally stop by a restaurant or tavern during the day trips
- If you have special requirements regarding the food (vegetarian, vegan, gluten-free, lactose-free, ...), please **let us know as soon as possible**

senier, 2017-12-07:
Nice location. Count me in!

m-stein, 2017-12-07:
@senier Nice :)

Kensan, 2018-01-19:
If there's still an open spot I would like to attend again this year.

m-stein, 2018-03-13:
### Update: Date and Accommodation

As there were unforeseeable issues with the Forsthaus Bielatal we recently had to switch to another accomodation. Sorry, for this inconvenience. This doesn't change the number of people that can participate.

#### Location

Trixie Holiday Park, Jonsdorfer Str. 40, 02779 Großschönau
- [Homepage](https://www.trixi-park.de/en/Trixi-Ferienpark/)
- [Open Street Maps](https://www.google.de/maps/place/Trixi+Holiday+Park/@50.878803,14.6727129,15z/data=!4m5!3m4!1s0x0:0xa08504f11e4e8518!8m2!3d50.878803!4d14.6727129)

#### Time
- Check-in as of 3 pm on May 25th (has not changed)
- Check-out till 10 am on May 28th

#### Room
- 6 vacation appartments (each with one two-bed room and one single-bed room) for 16 participants
- A self-catering kitchen in each appartment
- A lockable seminar room (has not changed)
- All exclusively used by us during the Hack'n'Hike (has not changed)

#### Misc
- Internet will be available
- Bedding and towels will be provided (has not changed)
- You can park your car at the site (has not changed)

#### Catering
- Genode Labs will provide catering for breakfast and dinner (has not changed)
- For lunch, we normally stop by a restaurant or tavern during the day trips (has not changed)
- If you have special requirements regarding the food (vegetarian, vegan, gluten-free, lactose-free, ...), please **let us know as soon as possible** (has not changed)

gwitmond, 2018-03-13:
Hi Martin,

If there is space left, I'd love to come over again. (This time I don't 
bail out.)

Cheers, Guido.

ehmry, 2018-03-14:
I will definitely be able to make this year.

m-stein, 2018-04-25:
### Journey Information
#### By train, Dresden main station - Großschönau
_25.05.2018_

Station | Time | Platform |  Train
-- | -- | -- | --
Dresden Hbf | dep 13:08 | 13 | TLX  RE2 (5255)
Zittau | arr 14:37 | 3b
Transfer time 4 min.
Zittau | dep 14:41 | 1a | TL    L7 (20918)
Großschönau(Sachs) | arr 14:53 | 3

_28.05.2018_

Station | Time | Platform |  Train
-- | -- | -- | --
Großschönau(Sachs) | dep 09:04 | 3 | TL    L7 (20909)
Zittau | arr 09:17 | 1a
Transfer time 4 min.
Zittau | dep 09:21 | 2a | TLX  RE2 (5250)
Dresden Hbf | arr 10:49 | 13

#### By train, Berlin Schönefeld Airport - Großschönau
_25.05.2018_

Station | Time | Platform |  Train
-- | -- | -- | --
Berlin-Schönefeld Flughafen | dep 10:45 | 11 | S     45
Berlin Südkreuz (S) | arr 11:18 | 11
walk 7 min.
Berlin Südkreuz | dep 11:26 | 3 | EC   379
Dresden-Neustadt | arr 13:00 | 7
Transfer time 15 min.
Dresden-Neustadt | dep 13:15 | 3 | TLX  RE2 (5255)
Zittau | arr 14:37 | 3b
Transfer time 4 min.
Zittau | dep 14:41 | 1a | TL    L7 (20918)
Großschönau(Sachs) | arr 14:53 | 3

_28.05.2018_

Station | Time | Platform |  Train
-- | -- | -- | --
Großschönau(Sachs) | dep 09:04 | 3 | TL    L7 (20909)
Zittau | arr 09:17 | 1a
Transfer time 45 min.
Zittau | dep 10:02 | 3a | OE    65 (62116)
Cottbus | arr 11:56 | 4
Transfer time 5 min.
Cottbus | dep 12:01 | 5 | RE     2 (63975)
Königs Wusterhausen | arr 12:48 | 1
Transfer time 15 min.
Königs Wusterhausen | dep 13:03 | 1 | RB    22 (18117)
Berlin-Schönefeld Flughafen | arr 13:17 | 4

chelmuth, 2018-04-25:
@m-stein thanks for the travel information but I doubt many of us will make the train at 9 on Monday. From past experience, we may leave the venue earliest at 11.

m-stein, 2018-04-25:
### Some activity suggestions
#### Hike to Jonsdorf's "city of rocks"
Distance: about 10 km in total
Duration: about 4 hours including a stop for lunch in the restaurant at the rock formation "Nonnenfelsen"
Pictures:
[image]
Restaurant at the rock formation "Nonnenfelsen"

[image]
Jonsdorf's "city of rocks"

#### Climbing Forest 
Location: next to our accomodation.
Description: provides several parcours from 1m height to 11m height with a variety of obstacles of different levels of difficulty to get over
Website: www.kletterwald-zittauergebirge.de
Pictures:
[image]
Afterwards we could have a short trip to one of the restaurants in Großschönau or barbecue directly in the climbing forest as offered on their website.

m-stein, 2018-04-25:
I've updated the journey information adding my offer for a drive-by-bike group and adding later options for the return journy by train around 12am instead of 9am.

trimpim, 2018-05-03:
@m-stein I would like to join the bike group. Do I need to fly in my bike or is there a possibility to rent one?

m-stein, 2018-05-03:
@trimpim Cool to hear that you'd like to join :)
I think that you don't need to bring your bike. You can have one of mine ([1](https://www.fahrwerk-dresden.de/bikes_detail.php?bikeid=332), [2](https://www.ternbicycles.com/bikes/472/verge-tour), [3](https://www.winora.com/en/GB/bikes/464/2018-santiago/?colorId=44&frame=3)) if you like or you can also rent one in the [RollOn Dresden](http://www.rollondresden.de/fahrraeder) (nice guys, I drove one week with one of their bikes).

m-stein, 2018-05-04:
The frame sizes are 1:61, 2:26, 3:56

m-stein, 2018-05-14:
Could anyone who would like to join in for the climbing forest please give a short feedback?

It's 18€ per person.
We start from our accomodation Sunday the 27.05. at 10:00 am towards the climbing forest (no long walk, it is located right behind the trixi park). At 10:30 am we'll have our introduction there. We can stay there for 2:30 h (i.e. until 1:00 pm; of course you can leave earlier if you want to), afterwards we look out for a place to have lunch and after this we may add a short hike till dinner.

nfeske, 2018-05-14:
@m-stein I'm in for the climbing! :-)

alex-ab, 2018-05-14:
@m-stein: I'm in.

Kensan, 2018-05-15:
I would like to join the climbing as well.

insidioussid, 2018-05-23:
@m-stein I just noticed this thread and read that we'd have to mention eating preferences as soon as possible.. I'm quite late. Sorry for that. However, I neither eat meat nor fish...

m-stein, 2018-05-24:
### A _rough_ schedule

#### Friday
**3 pm** arrival at the Trixi Park Großschönau
**evening** settle in and time for hacking
**dinner** buffet in the Trixi Park restaurant
#### Saturday
**breakfast** buffet in the Trixi Park restaurant
**10:30 am** hike to the Orgelsteig and the Nonnenfelsen near Jonsdorf
**1:30 pm** lunch in the inn at the Nonnenfelsen
**2:30 pm** hike back to the Trixi Park
**4:00 pm** time for hacking
**dinner** BBQ by Genode Labs
**night** time for hacking
#### Sunday
**breakfast** buffet in the Trixi Park restaurant
**10:00 am** short walk to the Climbing Forest
**10:30 am** introduction and climbing in the climbing forest
**1:30 pm** 20 minutes walk to the restaurant "Deutsche Eiche" in Großschönau
**2:00 pm** lunch in the restaurant
**evening** time for hacking
**dinner** Self-Catering by Genode Labs
**night** time for hacking
#### Monday
**breakfast** buffet in the Trixi Park restaurant
**10 am** end of the Hack'n'Hike

chelmuth, 2018-05-24:
Thanks @m-stein for the splendid program summary. I like to add only that the minibus leaves Dresden about 1 pm from our office address. We plan to have a quick lunch at about 12.

Kensan, 2018-05-24:
Apparently the bomb has (literally) been defused a couple of minutes ago:

https://twitter.com/PolizeiSachsen/status/999660782530023425

> On 24 May 2018, at 16:37, Martin Stein <notifications@github.com> wrote:
> 
> Another hint:
> Two days ago they've found a bomb from World War II in Dresden which they didn't manage to defuse until now. This is why at least until 6 pm of today the airport Dresden remains closed and there might be effects on the train schedules as well! Please keep yourself informed about the situation if you come from outside of Dresden. You can also ask us, so, we might help you finding an alternative.

m-stein, 2018-05-28:
Thanks for coming to Großschönau and having a great time with us guys! I hope you all had a good return journey?

I just started Issue #2841 for planning the next years Hack'n'Hike.

nfeske, 2018-05-29:
@m-stein I'd like to thank you for splendid job you did with organizing our Hack'n'Hike weekend! Given the massive trouble with the reservation of Forsthaus Bielatal, I am especially happy that the new location worked out so well in the end.

I enjoyed the weekend very much and already look forward to the next instance. :-)