genodelabs/genode #2413: fs_rom always returns valid sessions

ehmry, 2017-05-16:
``fs_rom`` will serve ``ROM`` for any request, regardless if there is a file present to back the session with. This is desirable when the file backing file will be created during the lifetime of a session, but in other cases it is better to deny a session so that a fallback goes into effect or so that a subsystem does not block indefinitely on a missing dependency.

ehmry, 2017-05-16:
23d5db80112815de5a7efb0d68067cf3dbaf923e adds a ``wait="..."`` option to ``fs_rom`` policies while retaining the present behavior as a default.

The result is that ``fs_rom`` can be used like this:
```xml
<start name="fs_rom">
	<!-- ... -->
	<config>
		<default-policy wait="no"/>
	</config>
</start>

<start name="lz_rom" ... />
	<!-- ... -->
</start>

<start name="rom_fallback">
	<!-- ... -->
	<config>
		<fallback/>
		<fallback label="fs"/>
		<fallback label="lz"/>
	</config>
	<route>
		<service name="ROM" label_prefix="fs ->" >
			<child name="fs_rom"/> </service>
		<service name="ROM" label_prefix="lz ->" >
			<child name="lz_rom"/> </service>
		<any-service> <parent/> </any-service>
	</route>
</start>
```

nfeske, 2017-05-17:
Admittedly, I am still struggling with the fallback use case as it smehow goes against the grain of our best practices: Telling components precisely what to do instead of having them probe for the availability of resources. I am wondering, is the use case of `rom_fallback` really strong enough to justify a configurable session-creation response in the error case? But this is just a side note.

In principle, I am open to discuss the response of `fs_rom` to session requests referring to unavailable files. But I don't like making this behavior a configurable policy. I ultimately think that the immediate response with `Service_denied` would be most natural. The current behavior is just there because files on a file system are not always updated atomically in place but sometimes unlinked before re-created (e.g., when saving a file in vim). So when timing is unfortunate, the `fs_rom` client may issue a session request during the intermediate state after unlink and before the new version exists.

If we find a way to fix this issue differently, we can relieve `fs_rom` from the current mechanism and implemente your suggested response. One idea would be to have a "bump-in-the-wire" VFS plugin (we may call it "transact") that sanitizes file updates by tweaking the semantics of `unlink` and `open`. Instead of forwarding unlink operations directly, it would keep the file but merely keep track of the fact that it is unlinked. The next time the file is opened for writing, the writer would get a truncated file. A reader would still obtain the previous consistent) version until the writer has closed the file. This "transact" VFS plugin would thereby solve the issue by making file updates transactional. Once in place, we could strip the weird blocking handling from `fs_rom`.

I would very much prefer this approach for two reasons:

* It would fix the atomic file update issue at the right leve, which is the file-system level. In fact, currently, the `fs_rom` tweak does not even work reliably so that we routinely have to employ a patch like 4b070ff071b3ae327e67c764e3405616c70b12d2. Hence, I would love to have clean solution.
* It would simplify `fs_rom` instead of making it more complicated (by introducing the hard-to-explain `wait` policy).

@ehmry What do you think?

ehmry, 2017-05-17:
@nfeske Yes, I would prefer ``fs_rom`` had a fixed behavior and would just reject sessions. I hadn't considered this Vim issue before and I would have to think about it. Other solutions that come to mind are a script in Noux to replace files with a move operation or maybe a specialized ``fs_config_rom``.

As for ``rom_fallback`` I think of it more as a hierarchy than probing. The original use was to improve Noux performance by keeping the most commonly used libraries in kernel memory and others on disk. Unfortunately it is the late 'long-shot' routes that are most apparent when looking at the config.

nfeske, 2017-06-29:
In the past few days, I experimented with the combination of ram_fs, fs_rom, and fs_report as a more flexible alternative to report_rom. In the process, I adjusted the fs_report.run test in this respect (https://github.com/genodelabs/genode/commit/5c24516abf80729187266e112f579ac6669813fd). In this scenario, and also in my forthcoming next Genode-desktop scenario, I use this combination to great success. The latter scenario, however, inherently depends on fs_rom's behavior to always accept ROM session even if no file is available yet.

In the concrete case, I have a driver subsystem that spawns drivers on demand depending on the devices present in the machine. The drivers run in a dynamic init instance. At integration time, it is unknown which driver instances will run and how they are named. Each of the drivers reports its findings (i.e., the found block devices) to a fs_report, which in turn writes the reports to the ram_fs. The driver manager requests these reports via fs_rom from the ram_fs. The driver manager knows what to ask for because it has defined the names of the driver instances. But I can not know the point in time when the driver's corresponding reports are actually available. Hence, it requests a ROM with the known name and gets a notification once the report comes in. It is actually a very cool pattern that I intend to use for the interactive desktop part of the system as well.

In short, in contrast to what I stated above, the current behavior of fs_rom accommodates my scenario quite perfectly. It does not just cover up the vim issue. ;-)

ehmry, 2017-06-29:
I will close this issue then. As package management matures then missing ROM modules will be less of an issue.

closed 2017-06-29 by ehmry