genodelabs/genode #2417: Part_blk: added partitions reporting

BorisMulder-CSL, 2017-05-18:
During the Hack n' hike 2017 I worked on a feature in part_blk that enabled it to report the partitions.
The commit for it can be found at: https://github.com/nlcsl/genode/tree/part_blk_report
Feel free to make comments and changes, and use it in any way you want.

Boris

chelmuth, 2017-05-22:
@ehmry Sorry, I was not aware that you implemented this feature in 24ee00c02b8aed30a08c21844a784e83ce33e563 already. Do you have something to add to @BorisMulder-CSL work?

chelmuth, 2017-05-22:
The topic branch refers to #2226.

ehmry, 2017-05-22:
@chelmuth I had an issue with bad checksums on the primary header and good checksums on the secondary and wanted a fallback feature, but that is it. I say move ahead with this issue.

chelmuth, 2017-05-22:
This is the second time I hear about checksum errors - are you also referring to GPT? I couldn't reproduce the issue and was wondering what is causing this.

ehmry, 2017-05-22:
Yes, this was a problem with tables created by ``sgdisk`` but I am not able to reproduce it, so it must have been fixed.

chelmuth, 2017-05-23:
@rite do you still experience `Error: Wrong GPT header checksum`? Did you use older versions of sgdisk?

rite, 2017-05-23:
I noticed this error a few months ago but couldn't reproduce it either. I guess the partition table was created by gparted on Ubuntu 14.04... Newer saw this error with partitions created with sgdisk on Ubuntu 16.04.02 LTS.

chelmuth, 2017-05-23:
With ca2243c, I added my AHCI port report feature to staging.

nfeske, 2017-06-06:
Included in Genode 17.05

closed 2017-06-06 by nfeske

ehmry, 2017-06-06:
There is a small issue that I have come across, the MBR partition types are printed in decimal, but hexadecimal representation is more common.

chelmuth, 2017-06-07:
@ehmry if you're referring to the report, this is pretty clear IMO as the numbers are not prefixed with `0x` which leaves them in decimal representation. But, I wouldn't mind to report hex numbers incl. the prefix.