genodelabs/genode #2422: server/fs_report

ehmry, 2017-05-20:
I want to try and visualize the init reports so I probably need to access reports via the file-system.

ehmry, 2017-06-09:
Rebased against 17.05, added a src recipe.

nfeske, 2017-06-12:
Very cool! This is exactly what I needed as well. ;-)

nfeske, 2017-06-15:
This works very well in my current scenario, except for one minor issue: The server always appends ".report" as suffix.

In my current scenario, I combine it with ram_fs and fs_rom to supply a configuration to a dynamic init instance. With the current built-in-policy, the configuration ends up being named "dynamic.config.report", which is a bit awkward. @ehmry Would you consider to remove the suffix? One can still easily append a suffix where it makes sense by using the label-rewriting mechanism on init.

ehmry, 2017-06-16:
The reason for the suffix was the same as ``fs_log``, if ``a -> b`` reports to ``/a/b``, then ``a -> b -> c`` will have problems reporting to ``/a/b/c``.  But unlike logs, not every member of the hierarchy will report and those that do append a label, so it should be safe to drop the suffix.

ehmry, 2017-06-16:
Fixed at 4da79e6e9769f84c512c1b35c3907e12279ef9f0

nfeske, 2017-06-20:
Cool! Thanks @ehmry :-)

commit c6bf3c35 (2017-06-12): server/fs_report: write reports to file-systems

Fix #2422
commit 47c616ab (2017-06-19): server/fs_report: write reports to file-systems

Fix #2422
closed 2017-06-29 by nfeske