genodelabs/genode #2423: RISC-V 1.9 and toolchain update

ssumpf, 2017-05-22:
Update base hw RISC-V implementation to ISA 1.9.1 and bump GCC version to 6.3

ssumpf, 2017-05-22:
@chelmuth: Please merge 41a4286 to staging.

chelmuth, 2017-05-22:
@ssumpf please rebase on current staging. I don't feel prepared to resolve the merge conflicts in the patch files.

ssumpf, 2017-05-23:
@chelmuth: Okay updated with be9f934 to current staging
@cproc: After installing GNAT 4.9.3 I still get this error: 
`configuring bootstrap gcc...
configure: error: GNAT is required to build ada
make: *** [tool_chain:428: build/bootstrap/gcc/Makefile] Error 1`
... what do I miss?

cproc, 2017-05-23:
@ssumpf: no idea, but 'build/bootstrap/gcc/config.log' might have a clue

ssumpf, 2017-05-23:
@cproc: Thanks, it seems I need `dev-libs/isl` version 0.15 or 0.16
`configure:6051: checking for isl 0.16 or 0.15`
but I still get the same error.

ssumpf, 2017-05-23:
@cproc: Here the log file, I cannot see what is wrong here: 

`g++ (Gentoo 4.9.3 p1.2, pie-0.6.3) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:4966: $? = 0
configure:4955: g++ -v >&5
Using built-in specs.
COLLECT_GCC=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.3/g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/4.9.3/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-4.9.3/work/gcc-4.9.3/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.3 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/include/g++-v4 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/python --enable-languages=c,c++,java,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 4.9.3 p1.2, pie-0.6.3' --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --enable-lto --without-cloog --enable-libsanitizer
Thread model: posix
gcc version 4.9.3 (Gentoo 4.9.3 p1.2, pie-0.6.3) 
configure:4966: $? = 0
configure:4955: g++ -V >&5
g++: error: unrecognized command line option '-V'
g++: fatal error: no input files
compilation terminated.
configure:4966: $? = 1
configure:4955: g++ -qversion >&5
g++: error: unrecognized command line option '-qversion'
g++: fatal error: no input files
compilation terminated.
configure:4966: $? = 1
configure:4970: checking whether we are using the GNU C++ compiler
configure:4989: g++ -c   conftest.cpp >&5
configure:4989: $? = 0
configure:4998: result: yes
configure:5007: checking whether g++ accepts -g
configure:5027: g++ -c -g  conftest.cpp >&5
configure:5027: $? = 0
configure:5068: result: yes
configure:5117: checking whether g++ accepts -static-libstdc++ -static-libgcc
configure:5134: g++ -o conftest -g -O2   -static-libstdc++ -static-libgcc conftest.cpp  >&5
configure:5134: $? = 0
configure:5135: result: yes
configure:5199: checking for gnatbind
configure:5215: found /usr/x86_64-pc-linux-gnu/gnat-gcc-bin/4.9/gnatbind
configure:5226: result: gnatbind
configure:5291: checking for gnatmake
configure:5307: found /usr/x86_64-pc-linux-gnu/gnat-gcc-bin/4.9/gnatmake
configure:5318: result: gnatmake
configure:5340: checking whether compiler driver understands Ada
configure:5363: result: no
configure:5372: checking how to compare bootstrapped objects
configure:5397: result: cmp --ignore-initial=16 $$f1 $$f2
configure:5457: checking for objdir
configure:5472: result: .libs
configure:5646: checking for the correct version of gmp.h
configure:5666: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5666: $? = 0
configure:5684: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5684: $? = 0
configure:5685: result: yes
configure:5701: checking for the correct version of mpfr.h
configure:5719: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5719: $? = 0
configure:5736: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5736: $? = 0
configure:5737: result: yes
configure:5754: checking for the correct version of mpc.h
configure:5771: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5771: $? = 0
configure:5787: gcc -c -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include   conftest.c >&5
configure:5787: $? = 0
configure:5788: result: yes
configure:5806: checking for the correct version of the gmp/mpfr/mpc libraries
configure:5837: gcc -o conftest -g -O2 -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include    conftest.c  -L/home/cbass/src/genode/genode.git/tool/build/lib-in [... cut ...]
configure:6064: gcc -o conftest -g -O2  -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include -I/home/cbass/src/genode/genode.git/tool/build/lib-install/include     -lisl -L/home/cbass/src/genode/genode.git/tool/build/lib-install/lib -L/home/cbass/src/genode/genode.git/tool/build/lib-install/lib -L/home/cbass/src/genode/genode.git/tool/build/lib-install/lib -lmpc -lmpfr -lgmp conftest.c  -lisl -lgmp >&5
configure:6064: $? = 0
configure:6071: result: yes
configure:6388: error: GNAT is required to build ada`

cproc, 2017-05-23:
@ssumpf: this seems to be the problem:

```
configure:5340: checking whether compiler driver understands Ada
configure:5363: result: no
```

So, you'd need to somehow configure Ada support into your host GCC.

ssumpf, 2017-05-23:
@cproc: Ack & will do, thanks again!

ssumpf, 2017-05-24:
@chelmuth: GCC 6.3.0 builds right now. I had to disable ADA, though. RISC-V and ADA just don't go their way. Right now I wrestle with base-hw, `put_char` is not my friend. Sorry!

ssumpf, 2017-08-24:
@chelmuth: if there still is time please add: 2d03c121d 91c0cf56f f2bc54d 77db289 to staging

ssumpf, 2017-08-25:
@chelmuth: + cf0c69c

nfeske, 2018-03-01:
As far as I am concerned, we can close this issue. The only known outstanding feature is support for Ada on RISC-V. Should this feature become of interest, we should open a dedicated issue.

closed 2018-03-01 by nfeske