genodelabs/genode #2438: depot: binary archives for libraries cannot be published

nfeske, 2017-06-06:
When trying to publish binary archives of the `backdrop` pkg recipe, the libpng, libc, and libz cannot be found in the depot. The `./tool/depot/dependencies` produces the following output:

```
~/s/genode> ./tool/depot/dependencies genodelabs/pkg/x86_32/backdrop-2017-06-06
Error: incomplete or missing archives:
       genodelabs/bin/x86_32/libc-2017-06-06
       genodelabs/bin/x86_32/libpng-2017-05-31
       genodelabs/bin/x86_32/zlib-2017-05-31
make: *** [checked_completeness] Error 1
```

The problem seems to be the install location of the build results of libraries that are - in contrast to regular targets - hosted in a subdirectory of the library's API. The `dependencies` tool (which is internally used by the publish tool) fails to distinguish both cases.

NobodyIII, 2017-06-07:
I ran into this issue too. The issue seems to be with the version numbers being inconsistent. For example, src/zlib-2017-05-31 compiles to bin/.../zlib-2017-04-24. It's the same for libpng.

nfeske, 2017-06-12:
Commit 718a0b012ca1fa87bfef32e0b893640525897d5f addresses the problem.

commit 718a0b01 (2017-06-12): depot: fix binary-archive handling for libs

The depot tool did not consistently distinct libraries from regular
targets when handling binary archives. The binary archive of a regular
target is located at <user>/<bin>/<arch>/<name> whereas a library is
located at <user>/<bin>/<arch>/<api>/<name>.

Fixes #2438
commit c8e4d271 (2017-06-19): depot: fix binary-archive handling for libs

The depot tool did not consistently distinct libraries from regular
targets when handling binary archives. The binary archive of a regular
target is located at <user>/<bin>/<arch>/<name> whereas a library is
located at <user>/<bin>/<arch>/<api>/<name>.

Fixes #2438
closed 2017-06-29 by nfeske