genodelabs/genode #2442: stdcxx: std::mutex unsupported

senier, 2017-06-10:
# Problem

I'm porting C++ code that makes use of std::mutex. Right now, this is  unsupported by Genodes stdcxx. As a results I get errors like:

```
error: ‘mutex’ is not a member of ‘std’
```
See senier/genode@366d31f for my current idea how to solve this. I'm not sure whether this a viable solution, though. Especially always linking to libgcc_eh seems questionable. Feedback is appreciated. 

# Background

Mutexes are enabled only if `_GLIBCXX_HAS_GTHREADS` is defined in `repos/libports/include/stdcxx/bits/c++config.h`. This results in various undefined thread-related types:

```
...
error: ‘__gthread_t’ does not name a type
...
```

Those types are not defined in the thread interface used by Genodes stdcxx right now (gthr-single.h). Enabling POSIX threads by linking to gthr-posix.h, however, yields a number of undefined references itself:

```
stdcxx.lib.so: undefined reference to `typeinfo for void'
stdcxx.lib.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr()'
stdcxx.lib.so: undefined reference to `std::__exception_ptr::exception_ptr::~exception_ptr()'
```

This is resolved by adding various exception handling related files to stdcxx:

```
SRC_CC  += eh_ptr.cc eh_unex_handler.cc eh_terminate.cc eh_term_handler.cc vterminate.cc fundamental_type_info.cc
```

That leaves use with one undefined reference:

```
stdcxx.lib.so: undefined reference to `_Unwind_RaiseException'
```

This symbol is defined in libgcc_eh.a. When adding this library in lib.mk, the adapted stdcxx test builds and runs:

```
Genode 16.11-822-g5e3e80734 <local changes>
17592186044415 MiB RAM and 8998 caps assigned to init
[init -> test-stdcxx] ° °° °°° test-stdcxx started °°° °° °
[init -> test-stdcxx] 2015.000000 - 5.000000 - 4
[init -> test-stdcxx] 2015 - 5 - 4
[init -> test-stdcxx] 123
[init -> test-stdcxx] caught std::invalid_argument
[init -> test-stdcxx] ° °° °°° test-stdcxx finished °°° °° °
Run script execution successful.
```

chelmuth, 2017-06-12:
From our experiences with *gthr-posix* and friends, I'd like to remark that this path needs thorough testing as it may have far reaching side effects among stdcxx, supc++ and the GCC libraries. Therefore, we disabled threading support in the tool chain as much as possible. There are also traces in the sources of specific issues (e.g., https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/cxx/unwind.c#L6). In the example case, we learned that it's no solution to link *libgcc_eh.a* to all shared libraries.

In my opinion, the first question to answer: Is it actually possible to build a *gthr-single* tool chain (incl. libraries and header files) and use it for a *gthr-posix* stdcxx variant.

senier, 2018-01-25:
We went down this path a little and had to realize that it is very stony and unpleasant. We created a custom toolchain with gthr-posix enabled and tried to make Genode build and run with it. This ended in a lengthy debug session revealing that the toolchain added undefined references to `pthread_cancel` and `pthread_once` into `ld.lib.so`. As a result, it crashed when relocating itself during initialization. As this happens before logging is initialized, tracking this down was a lot of fun. Not.

We then evaluated options to resolve this. While we could have mocked the missing symbols, this sounded hackish and wrong. Increasing the TCB for all components by a dependency on `posix` (and thus `libc`) was unacceptable anyways. We briefly looked into implementing a custom `gthr-genode` thread model, but that seemed awkward, too. Thread models in gcc are hardly documented and existing implementation almost always are thin bindings to native posix implementations (except, maybe Windows). Given the semantic mismatch between POSIX and Genode's native threading we abandoned this idea rather quickly.

Eventually, we ended up implementing the threading and synchronization parts of libcxx in a [custom library](https://github.com/Componolit/componolit/tree/cpp_concurrency/include/cxxthreads) using Genode mechanisms directly. This works out fairly well, so far. We do not need a custom toolchain and the TCB of single-threaded applications is not increased. Plus, the library can be extended as needed. 

Thread-local storage is not implemented right now. However, even the `gth-single` toolchain supports the `__thread` keyword and generates a call into `__emutls_get_address`. Currently we thrown an exception, but maybe TLS can be realized that way.

IMO, we can close this issue.

chelmuth, 2018-01-25:
I had a quick look at your implementation. What I like about it is its simplicity a property it shares with our pthread implementation. The unfortunate part in my opinion is that it does not use pthreads but complements them. That's the reason for the extra effort to access the Genode environment etc. Wouldn't it be better to use the libc/pthread layer in the STL instead of bypassing it?

cproc, 2020-04-25:
Commit f6843fb enables thread-related features by defining _GLIBCXX_HAS_GTHREADS, using gthr-posix.h in stdcxx and adding the undefined supc++ symbols to the ld/libc symbol files. Only tried with qtwebengine so far, where some of the features are used occasionally, but not stressed.

chelmuth, 2021-02-10:
According to #3925 `std::mutex` is not on the *unexpected failures list*. Can we close this issue?

cproc, 2021-02-11:
@chelmuth: it appears to be working so far, so I guess we can close the issue for now

closed 2021-02-11 by cproc