genodelabs/genode #2446: depot recipes

ehmry, 2017-06-12:
I'd like to start pushing recipes to the world repository, but there are API libraries missing in the main repo.

chelmuth, 2017-06-13:
@ehmry did you use *tool/abi_symbols* to generate the symbol files? If the tool is applied to *stdcxx.lib.so* it generates only ~670 lines in the symbol file compared to your version with ~4900 lines. I did not check if the ABI still works with more complex programs using *stdcxx*, so there may be issues. But, I think we should strip the symbol files as much as possible to our current use cases currently. Also, I'm wondering why symbols like `acosf` are exported by *stdcxx*.

ehmry, 2017-06-13:
@chelmuth no, I didn't know about the tool, I took the full list and removed things by prefix. But the tool is definitely taking out too much. Its bizarre how many permutations of string handling functions there are.

ehmry, 2017-07-04:
At 61f5e7e523728568c34d1bd6b6436f15304744be I have recipes for fs_rom, ram_blk, and vfs servers.

ehmry, 2017-08-31:
Terminal recipe at 5d29f8046faece58a556441cb4fab0ca85be5017

ehmry, 2018-02-05:
I had a problem with the stdcxx recipe defining `content` dependencies twice, fix at bf4cc4bfa22dd06853c23110a156d663e2f70e8d.

nfeske, 2018-02-06:
@ehmry I guess this issue is fixed on staging already (see https://github.com/genodelabs/genode/commit/65a830d01a8caaec156cc5cccaacacfd1d048f8c), or isn't it?

ehmry, 2018-02-06:
Yes, you are right, I just caught this bug on `sculpt`.

nfeske, 2018-03-01:
I close this generic issue as we surpassed the initial phase of depot usage.

closed 2018-03-01 by nfeske