genodelabs/genode #2452: init: child exit leaves dangling services

ehmry, 2017-06-19:
Consider the case of a component waiting on a driver to initialize and announce a service. If the driver fails to initialize and exits then any component waiting for the driver service should suffer ``Service_denied`` and continue without the service or exit, rather than wait indefinitely. This is most obvious when running scenarios on linux that require the alsa driver, but the driver cannot obtain exclusive access to the device.

chelmuth, 2017-06-21:
@nfeske are you fine with this patch?

ehmry, 2017-06-21:
There is the matter of propagating the discontinuation up to init parent, but I figure that can be another issue.

nfeske, 2017-06-22:
I am not confident that the approach of destroying the services of the child will work in all cases. The session objects referring to those services still exist and point to those service objects. Hence, we end up with dangling pointers.

Another problem is that service objects are created whenever a new configuration is imported. With the patch, if a server exited, its services would be destroyed on the event of the exit. However when importing the next config (apply_config) new service objects will be created because they appear in the config. To sum it up, the service objects should mirror the configuration, not runtime-dependent state. So the approach goes against the grain of init's design.

A better way to reflect a server exit as a `Service_denied` error would be to change the state of all sessions that are targeting the exited server but are not established yet. We would not destroy any object but merely change the session state to `SERVICE_DENIED` and wake up the client.

nfeske, 2017-06-22:
As a further thought, I am not convinced of the particular feature or rather its direction. When following this direction further, we cultivate a design pattern where clients probe for services. This is not what I want because it adds complexity in clients that is almost never tested. Instead, we should always tell the client what to do via its configuration. If the client is configured to open an audio connection, it should assume that functional an audio service is available. If this assumption is violated by the component's parent, it is not the job of the component to intelligently respond to it. It is the parent's problem.

As a general rule of thumb, whenever we don't know how to deal with unexpected conditions, let the parent decide by making the behavior configurable. In the concrete case, if a component should be usable with and without audio support, we should tell it whether to use audio or not in its configuration.

Another approach to solve the concrete case of the lacking alsa support would be to let the driver tolerate the absence of an audio interface, complain with an error message, and just stay around as a dummy service eating the audio data without exiting.

ehmry, 2017-06-24:
I had probing in mind but I can see that support would be inconsistent across components at best, so you've talked me out of it. I would still like to patch init to make it explicit when a service cannot be resolved in this case.

ehmry, 2018-06-04:
Not an issue for me anymore, closing.

closed 2018-06-04 by ehmry