genodelabs/genode #2458: libc: Opening a symlink with O_CREAT|O_NOFOLLOW results in infinte loop

senier, 2017-07-02:
See https://github.com/Componolit/componolit/commit/f8ad9d55f2c54166b05f438392ce6cdcabb3d22c for a minimal failing example.

senier, 2017-07-02:
Here is how I'd fix that: https://github.com/senier/genode/commit/92cd3b30279878def86b53bff7490eac26bf4d75

I'm not sure, though, whether testing for O_NOFOLLOW is sufficient to determine whether an ELOOP error is appropriate.

chelmuth, 2017-07-05:
I merged your fix but was wondering if it should adhere to the man page of `open`, which states

> ELOOP  pathname was a symbolic link, and flags specified O_NOFOLLOW but not O_PATH.

Must admit that I've no idea about the semantics of `O_PATH`...

senier, 2017-07-05:
I looked into this. The semantics of O_PATH is fairly simple in FreeBSDs libc - it doesn't exist ;-) So I guess we are safe.

commit 56e1f082 (2017-07-05): libc: handle O_CREAT|O_NOFOLLOW  in open correctly

We return ELOOP if the file already exists on
open(...,O_CREAT|O_NOFOLLOW).

Fixes #2458
commit fe22c8ef (2017-07-31): libc: handle O_CREAT|O_NOFOLLOW  in open correctly

We return ELOOP if the file already exists on
open(...,O_CREAT|O_NOFOLLOW).

Fixes #2458
commit 87c19cb1 (2017-08-17): libc: handle O_CREAT|O_NOFOLLOW  in open correctly

We return ELOOP if the file already exists on
open(...,O_CREAT|O_NOFOLLOW).

Fixes #2458
closed 2017-07-05 by senier