genodelabs/genode #2460: blk_cache makes excessive RAM upgrade requests

ehmry, 2017-07-04:
Previously init would hand out its slack RAM to components on request, and blk_cache relied on this. Now that init does not do this automatically blk_cache makes repeated RAM resource requests without effect when the cache saturates. Without a log throttling mechanism in place this can bog down the entire system.

My solution to this is to use a ``Ram_session_guard`` under the cache allocator, but I ran into an issue with properly setting the cache quota when dealing with quota donations and RPC buffer allocations. The simplest fix I found was to clamp the guard quota when the backend allocator fails.

ehmry, 2017-07-04:
Fix at 288c35c0fde1f469e0e636b9d8cf214802ba803f, a8466ef816f0b11b4b64daa7bf4ce7489d475696 (ignore 506d26dc589264ce68e0f8dd0b49568d7e9948d7).

ehmry, 2017-08-18:
Rebaseing and splitting recipes to a separate commit...

chelmuth, 2017-09-29:
I've to admit that I lost track of this issue. @ehmry could you please give an update if it is still relevant?

ehmry, 2017-09-29:
Relevant, but should be deferred until a later date.

nfeske, 2021-02-26:
I'm closing this issue because it has become stale.

closed 2021-02-26 by nfeske