genodelabs/genode #2462: vfs: symlink content silently dropped when above 1024 bytes

senier, 2017-07-04:
When creating a symbolic link using symlink(), the remainder of the link target gets silently dropped when the length of the target exceeds 1024 bytes.

The expected behavior is

1. Return a 'Filename too long' error when link target exceeds some internal limit
2. Support a link length of ~4096 bytes (the typical PATH_MAX size on Linux)
3. Expose an API to retrieve the respective limit (like `getconf PATH_MAX /` on Linux)

See [this test](https://github.com/Componolit/componolit/tree/master/src/test/mfe/symlink_length) for a reproducer (run script [here](https://github.com/Componolit/componolit/blob/master/run/mfe/symlink_length.run)).

ehmry, 2017-07-10:
For what its worth the maximum support symlink length within the VFS is pegged at Vfs::MAX_PATH_LEN and the ``ram_fs`` symlink length is pegged at ``File_system::MAX_PATH_LEN``.
https://github.com/genodelabs/genode/blob/master/repos/os/include/vfs/types.h#L29

I will fix the libc to return an error when the link gets too long, but I am on vacation with intermittent network so it might be a few more days.

ehmry, 2017-07-12:
Fixed at 55c8ee2b3269a70c7f86fea75259af67758aa17d.

@senier as you find problems feel free to add tests to ``repos/libports/src/test/libc_vfs/main.cc``, this test is regularly run against multiple VFS plugins.

senier, 2017-07-23:
I tested again with the fix in https://github.com/genodelabs/genode/commit/55c8ee2b3269a70c7f86fea75259af67758aa17d and it resolves the issue of an undetected excessive symlink. 'File name too long' is now signaled as expected when the symlink target exceeds MAX_PATH_LEN. Thanks, @ehmry .
Thanks also for the pointer to the vfs test case. I will consider adding future tests there.

chelmuth, 2017-07-26:
@ehmry why do you overload the length return value with an error case? May it possibly be better to just throw an instance of `File_system_session::Name_too_long`?

ehmry, 2017-07-26:
Yes, but this error happens during async packet write, and there is no return path for an error type.

commit fc50b1cd (2017-07-27): Check for symlink target length errors

Check for symlink length errors at the VFS library and the ram_fs and
vfs servers.

Fix #2462
closed 2017-08-18 by chelmuth