genodelabs/genode #2465: libc: stat()ing path with non-directory component does not result in ENOTDIR

senier, 2017-07-23:
When stating a path which contains a component that is not directory (but, e.g. a regular file), stat() should return a ENOTDIR error. Genodes implementation erroneously returns ENOENT in this case.

[POSIX](http://pubs.opengroup.org/onlinepubs/9699919799/) says:

> These functions shall fail if:
[ENOENT]
A component of path does not name an existing file or path is an empty string.
[ENOTDIR]
A component of the path prefix names an existing file that is neither a directory nor a symbolic link to a directory, or the path argument contains at least one non- <slash> character and ends with one or more trailing <slash> characters and the last pathname component names an existing file that is neither a directory nor a symbolic link to a directory.

The Linux and FreeBSD man pages on stat(2) make comparable statements.

chelmuth, 2017-09-29:
Did you investigate further where this issue stems from? A fix in form of a patch is highly appreciated.

nfeske, 2021-02-26:
We don't observe this as a problem in our daily use cases. Hence, I'm closing this issue.

closed 2021-02-26 by nfeske