genodelabs/genode #2467: port IOzone filesystem benchmark tool

ehmry, 2017-07-27:


ehmry, 2017-08-15:
@chelmuth My ``issue2467_iozone`` branch contains IOzone tests for RAM, Rump, and FatFS VFS plugins. The tests are in ``port/run`` and write data into ``run/data/iozone``. The ``run/iozone_plot`` script will display a visual plot of the data it finds there.

In short, RAM is always fast, Rump has fast reads, and FatFS is too slow to compare against the other two.

Ref #2410

chelmuth, 2017-08-24:
I skimmed over your branch and like the feature. For a merge, it needs to be rebased to staging (e.g., f5fd7a8aa64f44ac7f35a1ee7a8cf3156be9f5b8 and sync()).

ehmry, 2017-08-24:
Rebase.

chelmuth, 2017-10-19:
Does this deserve an issue branch rebased on current master ee4ee6a8ac91592065cbf7a96872a95acb926b17?

ehmry, 2017-10-19:
I expect so, I will ping you when it is ready.

ehmry, 2018-07-05:
@m-stein I've rebased my `issue2467_iozone` branch against staging.

nfeske, 2021-02-26:
There is no current plan to work on this issue.

commit 1b6f5c9c (2017-10-02): izone data collection and ploting run scripts

The run/iozone_* scripts test different file-systems and
run/iozone_plot displays a comparison with Gnuplot.

Fix #2467
closed 2021-02-26 by nfeske