genodelabs/genode #2468: init: attempt to replenish more quota than withdrawn

BorisMulder-CSL, 2017-07-31:
For some components, when I start them up and stop them by changing init's config, core gives me the following error:

Warning: detach: no attachment at 0x180000
Error: attempt to replenish more quota than withdrawn

More strangely, when I see init's init_ram report, it will show a ram quota value that is increasing every time such a child exits. As if its ram quota is growing because the child has given him more quota back than the child should have access to. After a certain amount of starts and stops init even completely hangs.

I have experienced this issue when the child is virtualbox, and with fetchurl. For the latter, I have made a run script below that should reproduce this issue. It should be ready to run, except that for it to work the IP adresses in fetchurl and in nic_bridge should be changed to addresses inside your local network.

Can you reproduce this issue and perhaps make a fix?

`#
# Build
#

set build_components {
	core init drivers/timer drivers/nic server/ram_fs server/fs_rom app/rom_logger app/fetchurl
	server/report_rom server/nic_bridge server/dynamic_rom
}

source ${genode_dir}/repos/base/run/platform_drv.inc

append_platform_drv_build_components

build $build_components

create_boot_directory

#
# Generate config
#

append config {
<config>
	<parent-provides>
		<service name="ROM"/>
		<service name="CPU"/>
		<service name="PD"/>
		<service name="RM"/>
		<service name="RAM"/>
		<service name="LOG"/>
		<service name="IRQ"/>
		<service name="IO_MEM"/>
		<service name="IO_PORT"/>
	</parent-provides>
	<default-route>
		<any-service> <parent/> <any-child/> </any-service>
	</default-route>
	<default caps="100"/>

	<start name="timer">
		<resource name="RAM" quantum="1M"/>
		<provides><service name="Timer"/></provides>
	</start>

	<start name="dynamic_rom">
		<resource name="RAM" quantum="4M"/>
		<provides><service name="ROM"/></provides>
		<config verbose="yes">
			<rom name="init.config">
				<inline description="iteration 1">
					<config>
						<report delay_ms="1000" init_ram="yes"/>
						<parent-provides>
							<service name="ROM"/>
							<service name="CPU"/>
							<service name="PD"/>
							<service name="RM"/>
							<service name="RAM"/>
							<service name="LOG"/>
							<service name="Nic"/>
    						<service name="File_system"/>
    						<service name="Timer"/>
						</parent-provides>
						<default-route>
							<any-service> <parent/> </any-service>
						</default-route>
						<start name="fetchurl" caps="200">
					     	<resource name="RAM" quantum="8M"/>
					     	<config>
						     	<libc stdout="/dev/log" stderr="/dev/log"
						     		ip_addr="10.0.2.132"
									netmask="255.255.255.0"
    								gateway="10.0.2.2"
    							/>
						     	<vfs>
						     		<dir name="dev">
						     			<log/> <null/> </dir>
						     		<fs label="tmp"/>
						     	</vfs>;
						     	<fetch url="http://genode.org/about/index" path="/index"/>
					     	</config>
				     	</start>
					</config>
				</inline>
				<sleep milliseconds="10000" />
				<inline description="iteration 2">
					<config>
						<report delay_ms="1000" init_ram="yes"/>
						<parent-provides>
							<service name="ROM"/>
							<service name="CPU"/>
							<service name="PD"/>
							<service name="RM"/>
							<service name="RAM"/>
							<service name="LOG"/>
							<service name="Nic"/>
    						<service name="File_system"/>
    						<service name="Timer"/>
						</parent-provides>
					</config>
				</inline>
				<sleep milliseconds="2000" />
			</rom>
		</config>
	</start>

	<start name="fs_rom">
		<resource name="RAM" quantum="10M"/>
		<provides>
			<service name="ROM"/>
		</provides>
		<route>
			<service name="File_system"> 
				<child name="ram_fs"/>
			</service>
			<any-service> <parent/> <any-child/> </any-service>
		</route>
	</start>

	<start name="rom_logger">
        <resource name="RAM" quantum="1M"/>
        <config rom="file" />
        <route>
            <service name="ROM" label="file"> <child name="fs_rom" label="index"/> </service>
            <any-service> <parent/> </any-service>
        </route>
    </start>

	<start name="ram_fs">
		<resource name="RAM" quantum="10M"/>
		<provides>
			<service name="File_system"/>
		</provides>
		<config verbose="yes">
			<policy label_prefix="init -> fetchurl" root="/" writeable="yes"/>
			<policy label_prefix="fs_rom" root="/" writeable="no"/>
		</config>
	</start>

	<start name="nic_bridge">
		<resource name="RAM" quantum="10M"/>
		<provides><service name="Nic"/></provides>
		<config verbose="yes" mac="12:34:56:78:90:ab">
			<policy label_prefix="init -> fetchurl" ip_addr="10.0.2.132"/>
		</config>
		<route>
			<service name="Nic"> <child name="nic_drv"/> </service>
			<any-service> <parent/> </any-service>
		</route>
	</start>
	
	<start name="nic_drv">}
append_if [have_spec linux] config {
		<binary name="linux_nic_drv"/>
}
append config {
		<resource name="RAM" quantum="5M"/>
		<provides><service name="Nic"/></provides>
		<config>
			<libc stdout="/dev/log" stderr="/dev/log">
				<vfs> <dir name="dev"> <log/> </dir> </vfs>
			</libc>
		</config>
	</start>

	<start name="report_rom">
		<resource name="RAM" quantum="1M"/>
		<provides> <service name="Report"/> <service name="ROM"/> </provides>
		<config verbose="yes"/>
	</start>

	<start name="init" caps="2000">
		<binary name="init"/>
		<resource name="RAM" quantum="10M"/>
		<configfile name="init.config"/>
		<route>
			<service name="ROM" label="init.config"> <child name="dynamic_rom"/> </service>
			<service name="File_system"> <child name="ram_fs"/> </service>
			<service name="Nic"> <child name="nic_bridge"/> </service>
			<service name="Report"> <child name="report_rom"/> </service>
			<any-service> <parent/> <any-child/> </any-service>
		</route>
	</start>}

append_platform_drv_config

append config {
</config>}

install_config $config

#
# Boot modules
#

set boot_modules { 
	core 
	curl.lib.so
	dynamic_rom
	fetchurl
	fs_rom
	init 
	ld.lib.so 
	libc.lib.so 
	libcrypto.lib.so 
	libcsl.lib.so 
	libm.lib.so
	libssh.lib.so 
	libssl.lib.so 
	linux_nic_drv
	lwip.lib.so
	nic_drv
	nic_bridge
	pthread.lib.so
	ram_fs
	report_rom
	rom_logger
	timer
	zlib.lib.so 
}

append_platform_drv_boot_modules

build_boot_image $boot_modules

append qemu_args   -nographic  

run_genode_until forever

`

nfeske, 2017-08-01:
Thank you for the detailed bug report and the test case! Just to double check, have you the commits f278024e44411509913ca3af9337bd6e53a1b9a5 and f95bfddc09113a83d8b1f1c2faecaa41d957079f present on your branch? They were added to master after 17.05 and address a problem that sounds very similar to what you are describing.

BorisMulder-CSL, 2017-08-01:
Well, when I cherry pick those commits, core still gives this error.
However, the init report says init does not keep growing its RAM quota; instead the number remains constant. Init also still hangs after a while (though that might be a problem of my scenario).

nfeske, 2017-08-01:
Please do not let the message irritate you too much. I left it there for myself but it appears in legit situations as well.

But the init-hanging problem is irritating. The only situation where a hang can legitimately happen is when env sessions are routed to child services that are misbehaving. E.g., when routing the env ROM session for the binary or dynamic linker to an fs_rom instance, init has to interact with fs_rom to load the ELF binary. Should the fs_rom service hang or become unavailable, init hangs as well. Init should not hang under any other circumstances.

BorisMulder-CSL, 2017-08-01:
Just a question here: does every ROM session count as an environment session? So also the one for the config or some other rom module?
In my case it's the config that breaks it up. Init gets the config from a ram_fs, then some other component changes the file, and due to some race condition in maybe fs_rom things go wrong. I'm investigating this further.

nfeske, 2017-08-01:
With the so-called "env" sessions, I refer to the sessions immediately created and called by the parent at the child construction. Specifically, these are the `Env_connection` objects found in `base/child.h`. They include the ROM for the executable and the dynamic linker as well as the child's PD and CPU sessions. Technically, the child's LOG session is also an env session. But as this session is newer called by the parent, it cannot pose any hanging of the parent (init).

The child's config ROM is not an env session because it is initiated by the child.

I recently encountered an issue related to the dynamic use of ram_fs and fs_rom (while I tried to emulate report_rom with fs_report, fs_rom, and ram_fs). Sometimes, file changes at the ram_fs remained unnoticed at the fs_rom client. Can you please check if the following commits make a difference? 

* c1ea08e511a612ee136d8476846d0153792b5236
* 2cffc0e30784d677c44cf822aa14200b37a2d8f0
* 59378fac535c5ab72aa29356e8ad0e2831923818

The corner cases are now covered by a new test (5c24516abf80729187266e112f579ac6669813fd). Right now, the commits cited above are at our staging branch.

BorisMulder-CSL, 2017-08-03:
I tried cherry-picking those commits, but the fs_rom gives compiler errors now. Could it be that more commits are required that change something to fs_rom?

nfeske, 2017-08-04:
Yes, the preceding commit 4e4071766bb0565e2180b3ee46c21644a5c50f3f looks important (also, there are no other commits between master and the ones I cited).

BorisMulder-CSL, 2017-09-22:
With these new commits in 17.08, init's ram usage does not leak anymore. The warning and the error remain. If you think this is not a problem, this issue can be closed.

closed 2019-03-17 by nfeske