genodelabs/genode #2469: Ready-only policies not supported by all Block servers

ehmry, 2017-07-31:
In some situations (perhaps most if embedded Genode appliances proliferate) it is vital enforce read-only access to boot media. I suggest the ``Block::Driver`` class and all derived drivers optionally enforce ready-only ``Block`` sessions by policy.

ehmry, 2017-07-31:
I have ``writeable`` policy support in ``ahci_drv`` and ``part_blk`` at 5da41f3e124d1bad45b1c816ab1c0214cb242c21. ``usb_drv`` was non-trivial and uses a static ``Block`` root component so I left it as-is.

ehmry, 2017-08-18:
@chelmuth Rebased as read-only be default at bd45a761d8bf22a8da3c56f3a82b1c0e48f6e658.

chelmuth, 2017-08-23:
Some libc_vfs run script produce errors like
```
[init -> ahci_drv] read-only session opened at device 0 for 'vfs -> 0'
```
It is also worthwhile to remove *libm* and *posix* from the libc_vfs tests as the test component does not depend on them, i.e., the libraries are not built and may not be available for integration.

ehmry, 2017-08-23:
Rebased and fixed at d0b31187cc7cee42a4ddb8f2b04d284fcbc9654e

chelmuth, 2017-08-24:
`make run/libc_vfs_fat KERNEL=...` still produces the error above, but is not in the autopilot list. Therefore, I merged the commit.

commit d8164f0a (2017-08-24): default ahci_drv and part_blk Block sessions to read-only

Add a "writeable" policy option to the ahci_drv and part_blk Block
servers and default from writeable to ready-only. Should a policy
permit write acesss the session request argument "writeable" may still
downgrade a session to ready-only.

Fix #2469
closed 2017-08-30 by chelmuth