genodelabs/genode #2472: tool: exact build description for external binaries.

jklmnn, 2017-08-04:
For each  binary that is shipped with this repository there should be a description how it was built to be able to rebuild it.

E.g. I failed to rebuild the exact Bender that is used in Genode. The [README](https://github.com/genodelabs/genode/blob/master/tool/boot/README) only states the used repository but neither a commit id or revision nor any options that have been used. From this point it is impossible (except by brute forcing builds) to create a binary from the same source as it was used to create the one in this repository.

The descriptions for grub and tianocore are examples how this could be done.

alex-ab, 2017-08-04:
```
  The changes are available on the genode_bender branch of
  https://github.com/alex-ab/morbo.git.
```

The branch is mentioned.

I must admit, a exact commit it would be helpful.

For build instructions, read the README of the external repository (as you also has to do for GRUB ...).

It states to call 'scons'.

nfeske, 2017-08-04:
Thanks @jklmnn for the remark and @alex-ab for the quick fixup. I merged it to staging.

jklmnn, 2017-08-04:
Thanks for the quick response! I think I can close this.

closed 2017-08-04 by jklmnn