genodelabs/genode #2474: nanosleep ignores nanoseconds

senier, 2017-08-04:
When using nanosleep() I find that the nanoseconds portion of the timespec is ignored. Tests are performed on Nova/qemu.

```cpp
...
time_t last = time(nullptr);
struct timespec duration;   
duration.tv_sec = 3;
duration.tv_nsec = 500000000;
nanosleep(&duration, NULL);
double delta = difftime(time(NULL), last);
...
```

The value of `delta` is 3.0 contrary to the expected value of 3.5. For values below a second, there is no delay at all. The full test can be found [here](https://github.com/senier/componolit/tree/issue16/src/test/mfe/difftime_subsecond).

Occasionally, I get a warning which may or may not be related:

```
Warning: submitting signal failed - error 4 - context=cap<34548>
```

alex-ab, 2017-08-07:
From the man page:

```
The difftime() function returns the number of seconds elapsed between
time time1 and time time0, represented as a double.
```

With following patches:

```
--- a/repos/libports/src/lib/libc/nanosleep.cc
+++ b/repos/libports/src/lib/libc/nanosleep.cc
@@ -15,12 +15,15 @@
 #include <sys/time.h>
 
 #include "task.h"
+#include <base/log.h>
 
 extern "C" __attribute__((weak))
 int _nanosleep(const struct timespec *req, struct timespec *rem)
 {
        unsigned long sleep_ms = req->tv_sec*1000 + req->tv_nsec/1000000;
 
+       Genode::error("sleep_ms ", sleep_ms, " ", req->tv_sec, " ", req->tv_nsec);
+
        struct Check : Libc::Suspend_functor { bool suspend() override { return true; } } check;
        do { sleep_ms = Libc::suspend(check, sleep_ms); } while (sleep_ms);
 
```

```
--- a/src/test/mfe/difftime_subsecond/main.cc
+++ b/src/test/mfe/difftime_subsecond/main.cc
@@ -6,6 +6,7 @@
 #include <stdio.h>
 #include <unistd.h>
 #include <err.h>
+#include <time.h>
 
 /* stdcxx includes */
 #include <string>
@@ -25,10 +26,12 @@ void Libc::Component::construct(Libc::Env &env)
                duration.tv_nsec = 500000000;
                nanosleep(&duration, NULL);
 
-               double delta = difftime(time(NULL), last);
+               time_t x = time(NULL);
+
+               double delta = difftime(x, last);
                if (delta < 3.5)
                {
-                       Genode::log("Invalid time diff: ", delta);
+                       Genode::log("Invalid time diff: ", delta, " ", x, "-", last, " diff=", x - last);
                        exit (1);
                }
```

I get:

```
[init -> difftime_subsecond] Error: sleep_ms 3500 3 500000000
[init -> difftime_subsecond] Invalid time diff: 3.0 1502096943-1502096940 diff=3
[init] child "difftime_subsecond" exited with exit value 1
```

time() and difftime() operate on second granularity, so the both seem not to be the right functions for your use-case.

chelmuth, 2017-08-07:
@alex-ab is perfectly right accord. to the man page of `difftime`.

> Each of the times is specified in calendar time, which means its value is a measurement (in seconds) relative to the Epoch, 1970-01-01 00:00:00 +0000 (UTC).

So, sub-second granularity is impossible.

senier, 2017-08-08:
I can see that one could come to the conclusion that difftime has a one second granularity from the man page. But not necessarily, as the unit of measurement does not imply the same granularity. An why would they return the value as a double instead of an integer if it cannot be sub-second?

Anyways, I confirmed the same behavior on Linux and I guess the Genode implementation is in line with other implementations. Hence, not a bug - just a stupid API.

Thanks for looking into that!

closed 2017-08-08 by senier

chelmuth, 2017-08-15:
In this case, the API is specified in the ISO C standard http://pubs.opengroup.org/onlinepubs/009695399/functions/difftime.html and, yes, it feels awkward to express *number of seconds* as double.