genodelabs/genode #2481: riscv_spike platform: can't link hard-float modules with soft-float modules

mneilly, 2017-08-13:
I tried to compile and run the demo for RISC-V following the information on the following page:

https://genode.org/documentation/developer-resources/getting_started

Compilation fails due to a mixing for hard and soft float...

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working tree clean

$ /usr/local/genode-gcc/bin/riscv64-unknown-elf-gcc-6.3.0 --version
riscv64-unknown-elf-gcc-6.3.0 (GCC) 6.3.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ ./tool/create_builddir riscv_spike
Successfully created build directory at /home/eng/mneilly/OSS/genode/build/riscv_spike.
Please adjust /home/eng/mneilly/OSS/genode/build/riscv_spike/etc/build.conf according to your needs.

$ make -C build/riscv_spike run/demo
    ...
    LINK     launchpad
libs=/home/eng/mneilly/OSS/genode/build/riscv_spike/var/libcache; /usr/local/genode-gcc/bin/genode-riscv-g++ -Wl,-gc-sections -Wl,-z -Wl,max-page-size=0x1000 -Wl,--dynamic-l
ist=/home/eng/mneilly/OSS/genode/repos/base/src/ld/genode_dyn.dl -nostdlib -Wl,-nostdlib -Wl,-Ttext=0x01000000  -Wl,--dynamic-linker=ld.lib.so -Wl,--eh-frame-hdr -Wl,-rpath-
link=. -Wl,-T -Wl,/home/eng/mneilly/OSS/genode/repos/base/src/ld/genode_dyn.ld -Wl,--whole-archive -Wl,--start-group launcher.o launchpad_window.o main.o $libs/base/base.lib
.a $libs/blit/blit.lib.a $libs/launchpad/launchpad.lib.a $libs/scout_gfx/scout_gfx.lib.a $libs/scout_widgets/scout_widgets.lib.a ld.lib.so  -Wl,--no-whole-archive -Wl,--end-
group /usr/local/genode-gcc/bin/../lib/gcc/riscv64-unknown-elf/6.3.0/rv64imac/lp64/libgcc.a -o launchpad
/usr/local/genode-gcc/bin/../lib/gcc/riscv64-unknown-elf/6.3.0/../../../../riscv64-unknown-elf/bin/ld: /home/eng/mneilly/OSS/genode/build/riscv_spike/var/libcache/scout_widgets/scout_widgets.lib.a(binary_closed_icon.rgba.o): can't link hard-float modules with soft-float modules
```
As can be seen in the output, compilation fails with 

**> can't link hard-float modules with soft-float modules**

ssumpf, 2017-08-13:
RISC-V support is unfortunately and currently broken. The toolchain update to GCC 6.3 and the ISA update to 1.9 did not go very well. Sorry for that, it is on my agenda though.

chelmuth, 2017-08-14:
It's unfortunate that the build process breaks with an error message. But on the other hand, the demo run script never worked on RISC-V for many reasons with missing peripheral specification (and therefore missing drivers) the most important. We enabled only a small subset of Genode's feature set on this platform (incl. the dynamic linker). I'd suggest to not expect any tests to work smoothly beside `make run/log`.

nfeske, 2021-02-26:
With Genode 21.02, this issue is fixed.

closed 2021-02-26 by nfeske