genodelabs/genode #2484: nova: 'Rpc_entrypoint::_dissolve()' can hang

cproc, 2017-08-15:
When running the 'noux_bash' test on NOVA and entering the command

```for ((i=0;;i++)); do /bin/echo $i; done```

the test usually hangs in core in 'Rpc_entrypoint::_dissolve()' after some rounds.

More specifically, it appears to hang at: https://github.com/genodelabs/genode/blob/0d1be4abe2294155e36bc0eb0c1cd61499f08d93/repos/base-nova/src/lib/base/rpc_entrypoint.cc#L105

The problem occurs since commit 843dd179d7165853d77ecf175f330e5e4f51cbb4 and seems to be related to the added lock guard in 'Synchronized_ram_session::free()', which has moved to

https://github.com/genodelabs/genode/blob/0d1be4abe2294155e36bc0eb0c1cd61499f08d93/repos/base/src/core/include/synced_ram_allocator.h#L44

in the meantime.

cproc, 2017-08-15:
Added a test case with automatic execution of the shell command and debug messages.

alex-ab, 2017-08-16:
Thanks for the detailed analysis. The commit 97e04f4e1789fa833993c1174d53a6eaeda29546 let for me pass your test on qemu endless, however on hardware it gets stuck after a while. However the test log looks good, so there is probably another reason because it get stuck.

commit 29f708e4 (2017-08-18): core: serve parent and core service by one ep

Fixes #2484
closed 2017-08-30 by chelmuth