genodelabs/genode #2487: booting with bender causes core to fail to read config rom

jklmnn, 2017-08-17:
When booted with bender (and NOVA) on certain hardware, core fails to correctly parse the config and therefor init doesn't find any services to start end exists immediately.

# Bug description

This bug is bound to a specific image on certain hardware. This means an image containing this bug will trigger it only but always when booted on a specific device. Also it is no random hardware error as I could reproduce it on two devices of the same type. The same image booted on another type of device or in Qemu will work flawlessly.

I could track down the root of  the issue but I still can't specify an exact cause.
Inside the [`xml()`](https://github.com/genodelabs/genode/blob/master/repos/base/include/base/attached_rom_dataspace.h#L147) function, the memory address of the config rom is correct and the valid bit is set. When the bug occurs, an `Invalid_syntax` exception has been thrown and therefor an empty node is returned.
This exception is caused by the [`Xml_node`](https://github.com/genodelabs/genode/blob/master/repos/base/include/util/xml_node.h#L578) constructor. Here comes the weird part.
I printed the start tag. If it is of type `EMPTY` the function should return. To check that I casted the tag to int which results in 2 (which is correlates with the enum and is also what is printed if the bug is not triggered). The log message is printed directly before the checks and before the exception could be thrown (see [`Xml_node@646056e7e`](https://github.com/jklmnn/genode/blob/designware_linux_task_lxerr_2/repos/base/include/util/xml_node.h#L579)).
The log message is clearly before the exception but the exception is not preceded by these messages in the log. Contrary there is a message after each exception. Also none of the called constructors throws this exception.
I see two possibilities here:
 - either the log messages are in disorder and the exception gets thrown although the constructor should return before
 - or the exception gets thrown before the log message but I cannot find any other plausible source at this point.

I have observed this on two devices, a [Lenovo Yoga 300-11IBR](https://www.cyberport.de/lenovo-yoga-300-11ibr-2in1-notebook-schwarz-n3160-windows-10-1C31-30T_8274.html) and a [TrekStor SurfTab wintron 7.0](https://www.cyberport.de/trekstor-surftab-wintron-7-0-16-gb-wifi-win-10-schwarz-1H29-00A_1530.html). While I could trigger the bug on both devices I couldn't do that with a single image.

[images.zip](https://github.com/genodelabs/genode/files/1230985/images.zip)

The image.zip file contains the bender, nova and genode images I have used.
sha1sum:
```
b4278329d75d0347ff7c14dc47ddc8b7f2520e34  image.elf
2cee5a51bf5a7ce7ca5f344341562f85a61da034  image.elf.orig
c67b42f5e4f2d757e65ed8f35354cd8959981861  bender
564c5d2de6c9a60eff1f0b704048eba2bb7c1163  hypervisor
```

With these files I could create the following results:

 | Wintron | image.elf | image.elf.orig |
 | -- | -- | -- |
 | with bender | page fault | bug |
 | w/o bender | bug or page fault | normal |

 | Yoga | image.elf | image.elf.orig |
 | -- | -- | -- |
 | with bender | bug or page fault | normal |
 | w/o bender | normal | normal |

 | Qemu | image.elf | image.elf.orig |
 | -- | -- | -- |
 | with bender | normal | not tested |
 | w/o bender | normal | not tested |

Bug means in this case that the bug I described above occured. The page fault appeared on loading init before any services were started. In the bug or page fault cases I couldn't determine why it wouldn't boot as I had no other debugging options. I couldn't test the image.elf.orig since I wasn't able to whip up a qemu bootable image and I had only this image file left.

# Hardware

## TrekStor Wintron

### dmesg 

```
[    0.000000] Linux version 4.9.18+ (root@p2pram) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP Thu May 11 09:38:20 CEST 2017
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000037a76fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000037a77000-0x0000000037a7ffff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000037a80000-0x0000000037afffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000037b00000-0x000000003bb8dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000003bb8e000-0x000000003bdadfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003bdae000-0x000000003bdaefff] usable
[... 1389 lines cut ...]
cpu family	: 6
model		: 76
model name	: Intel(R) Celeron(R) CPU  N3160  @ 1.60GHz
stepping	: 4
microcode	: 0x408
cpu MHz		: 713.281
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 3200.00
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```

alex-ab, 2017-08-17:
Can you please provide or check the core boot information about the detected memory and where the loaded ROMs got placed to. Maybe there is something suspicious regarding the physical regions (compared to the BIOS-e820 physical ranges).

jklmnn, 2017-08-18:
I can only check this for the TrekStor tablet and Qemu since I don't have any debugging on the other device.

# TrekStor tablet
## image.elf with Bender (page fault)
```
Bender: Hello World.
NOVA Microhypervisor v7-2006635 (x86_64): Aug 17 2017 09:16:01 [gcc 6.3.0] [MBI2]

[ 0] TSC:2133280 kHz BUS:0 kHz DL
[ 0] CORE:0:0:0 6:37:8:1 [82f]         Intel(R) Atom(TM) CPU  Z3735G @ 1.33GHz
[ 3] CORE:0:3:0 6:37:8:1 [82f]         Intel(R) Atom(TM) CPU  Z3735G @ 1.33GHz
[ 2] CORE:0:2:0 6:37:8:1 [82f]         Intel(R) Atom(TM) CPU  Z3735G @ 1.33GHz
[ 1] CORE:0:1:0 6:37:8:1 [82f]         Intel(R) Atom(TM) CPU  Z3735G @ 1.33GHz
Hypervisor features VMX
Hypervisor reports 4x1 CPUs
CPU ID (genode->kernel:package:core:thread) remapping
 remap (0->0:0:0:0) boot cpu
 remap (1->1:0:1:0)
 remap (2->2:0:2:0)
 remap (3->3:0:3:0)
Hypervisor info page contains 30 memory descriptors:
core     image  [0000000000100000,0000000000787000)
binaries region [0000000000226000,0000000000787000) free for reuse
detected physical memory: 0x0000000000000000 - size: 0x000000000008f000
use      physical memory: 0x0000000000000000 - size: 0x000000000008f000
detected physical memory: 0x0000000000090000 - size: 0x000000000000e000
use      physical memory: 0x0000000000090000 - size: 0x000000000000e000
detected physical memory: 0x0000000000100000 - size: 0x000000001ff00000
use      physical memory: 0x0000000000100000 - size: 0x000000001ff00000
detected physical memory: 0x0000000020200000 - size: 0x0000000017877000
use      physical memory: 0x0000000020200000 - size: 0x0000000017877000
detected physical memory: 0x0000000037b00000 - size: 0x000000000408e000
use      physical memory: 0x0000000037b00000 - size: 0x000000000408e000
detected physical memory: 0x000000003bdae000 - size: 0x0000000000001000
use      physical memory: 0x000000003bdae000 - size: 0x0000000000001000
detected physical memory: 0x000000003bdb0000 - size: 0x0000000000002000
use      physical memory: 0x000000003bdb0000 - size: 0x0000000000002000
detected physical memory: 0x000000003bdb3000 - size: 0x0000000000a4d000
use      physical memory: 0x000000003bdb3000 - size: 0x0000000000a4d000
detected physical memory: 0x000000003cc00000 - size: 0x00000000000c7000
use      physical memory: 0x000000003cc00000 - size: 0x00000000000c7000
detected physical memory: 0x000000003cdf8000 - size: 0x0000000000008000
use      physical memory: 0x000000003cdf8000 - size: 0x0000000000008000
:virt_alloc: Allocator 0x1e75f0 dump:
 Block: [0000000000002000,0000000000003000) size=4K avail=0 max_avail=0
 Block: [0000000000003000,0000000000004000) size=4K avail=0 max_avail=0
 Block: [0000000000004000,0000000000005000) size=4K avail=0 max_avail=0
 Block: [0000000000005000,0000000000006000) size=4K avail=0 max_avail=0
 Block: [0000000000006000,0000000000007000) size=4K avail=0 max_avail=0
 Block: [0000000000007000,0000000000008000) size=4K avail=0 max_avail=0
 Block: [0000000000008000,0000000000009000) size=4K avail=0 max_avail=0
 Block: [0000000000009000,000000000000a000) size=4K avail=0 max_avail=0
 Block: [000000000000a000,000000000000b000) size=4K avail=0 max_avail=137434760164K
 Block: [000000000000b000,000000000000c000) size=4K avail=0 max_avail=0
 Block: [000000000000c000,000000000000d000) size=4K avail=0 max_avail=0
 Block: [000000000000d000,000000000000e000) size=4K avail=0 max_avail=0
 Block: [000000000000e000,000000000000f000) size=4K avail=0 max_avail=0
 Block: [000000000000f000,0000000000010000) size=4K avail=0 max_avail=0
 Block: [0000000000010000,0000000000011000) size=4K avail=0 max_avail=0
 Block: [0000000000011000,0000000000012000) size=4K avail=0 max_avail=0
 Block: [0000000000012000,0000000000013000) size=4K avail=0 max_avail=0
 Block: [0000000000013000,0000000000014000) size=4K avail=0 max_avail=137434760164K
 Block: [0000000000014000,0000000000015000) size=4K avail=0 max_avail=0
 Block: [0000000000015000,0000000000016000) size=4K avail=0 max_avail=0
 Block: [0000000000016000,0000000000100000) size=936K avail=936K max_avail=936K
 Block: [0000000000226000,0000000000227000) size=4K avail=0 max_avail=0
 Block: [0000000000227000,0000000000228000) size=4K avail=0 max_avail=137434760164K
 Block: [0000000000228000,0000000000229000) size=4K avail=0 max_avail=0
 Block: [0000000000229000,00000000a0000000) size=2619228K avail=2619228K max_avail=2619228K
 Block: [00000000b0000000,00000000bfeff000) size=261116K avail=261116K max_avail=137434760164K
[... 538 lines cut ...]
[init] child "report_rom" announces service "Report"
[init] child "report_rom" announces service "ROM"
[init] child "nitpicker_config" announces service "ROM"
[init] child "acpi_report_rom" announces service "Report"
[init] child "acpi_report_rom" announces service "ROM"
[init] child "timer" announces service "Timer"
[init -> fb_boot_drv] Framebuffer with 800x600x32 @ 0x80000000
[init -> nitpicker_config] Warning: top-level node <xray> missing in input ROM xray
[init -> nitpicker_config] Warning: could not obtain input value for input xray_enabled
[init] child "fb_boot_drv" announces service "Framebuffer"
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init] child "platform_drv" announces service "Platform"
[init -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> ps2_drv] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> platform_drv] PS2 uses IRQ, vector 0x1
[init -> platform_drv] PS2 uses IRQ, vector 0xc
[init] child "ps2_drv" announces service "Input"
[init] child "nitpicker" announces service "Nitpicker"
[init -> scout] png is 101 x 92, depth=8
```

alex-ab, 2017-08-30:
The logs don't reveal for me anything suspicious. In #2242 the patch 346c9464220578f3dc295c773660a849d9738fc8 may potential be related to this issue. The framebuffer memory was not excluded/reserved from the normal available physical RAM. In principle, it may happen that core or init would use it as normal backing store, which would lead to any kind of corruption of the data structures placed in the framebuffer.

jklmnn, 2017-08-30:
I will try that fix and report if it solved the problem.

alex-ab, 2017-08-30:
I fear, bender needs also the same exclusion of framebuffer memory from the "free to use" list used for relocation. I'm currently investigating.

alex-ab, 2017-08-31:
@jklmnn: may you give bender from https://github.com/alex-ab/morbo/commits/genode_bender_overlap a try (and revert the DEBUG commit) ? It mainly should remove the framebuffer memory from the free available memory in bender. If the issue persists, please re-add the DEBUG commit and have a look to the output or provide here. Thanks!

jklmnn, 2017-09-11:
I have tried both the fixed bender with and without debug outputs. They all work fine in Qemu (as expected) but neither works on the tablet. I can't give you any debug output as they didn't boot at all (at least there was no output over serial).
They both booted on the yoga but gave the same results as the old bender. I can't give you debug output here either since the yoga doesn't have any serial interface.

alex-ab, 2017-10-20:
I had now something similar. In my case it turned out that the image.elf on the usb stick had another shasum then the one I actually generated - so the content was corrupted at the end.

chelmuth, 2017-10-20:
To ensure all data is written, I use the following commandline to write images to USB.
```
dd if=<image file> of=<device> bs=8M conv=fsync
```

alex-ab, 2017-11-14:
With commit e29b0fb70d5a58bad83f4a57ed048de71681c41 the issue should be fixed.

nfeske, 2017-11-14:
Merged to staging.

commit 23b16492 (2017-11-14): bender: fix free memory calculation for Multiboot2

Fixes #2487
commit d479e487 (2017-11-16): bender: fix free memory calculation for Multiboot2

Fixes #2487
commit 9646c17f (2017-11-30): bender: fix free memory calculation for Multiboot2

Fixes #2487
closed 2017-11-30 by chelmuth