genodelabs/genode #2490: virtualbox: network access from behind current nic_router

m-stein, 2017-08-17:


m-stein, 2017-08-17:
@chelmuth I created some (mostly unrelated) commits on my way to a working test scenario that may be interesting for staging.

m-stein, 2017-08-17:
6dd9e55 core: ignore boot modules with size 0
f15f522 core: generic ROM module initialization
3ff6ffd prepare_port sel4: check required python modules
cf7b068 core, region map: handle failing metadata assign
6649a4d allocator avl: exception if metadata assign fails
15bd3b0 core, region map: more descriptive variable name
b831c36 nic_router: error if uplink domain not configured

m-stein, 2017-08-29:
@chelmuth I just fixed a problem with a libc compile error when doing the NIC-router test and there are also some older commits left regarding this issue:
6bb6381 libc file operationcs: fix ambiguous size_t
702cdf0 nic_dump: link state and mac address
81d6a8b nic_dump: synchronize down/uplink creation

m-stein, 2017-08-29:
Ok, 6bb6381 actually is a fixup for 0615e2a.

m-stein, 2017-08-29:
Another fix for linking errors with libc variants:
2f63819 libc_resolv/_lxip: fix undefined references...

Could someone please have a short review on this? I assume it is related to 0615e2a too.

m-stein, 2017-08-30:
I did a quick implementation of a DHCP server in the NIC router. It enables me to do ``wget genode-labs.com`` from my tinycore vbox guest through the router acting as UDP/TCP NAT for vbox :)

m-stein, 2017-08-31:
@chelmuth For now two smaller fixes for the NIC router:
dbba45f nic_router: provide link state
c557b90 nic_router: use local but individual MAC addresses

m-stein, 2017-09-12:
@chelmuth Three commits that solve corner-case problems with timeouts:
ae05279 run/timeout: test if maximum timeouts trigger bugs
02d20ba timeout: fix outdated time in alarm scheduler
f8a6223 alarm: fix information loss due to int-cast

chelmuth, 2017-09-14:
59709cf5eb16de241270fa462e9af3a511ef0f49 breaks `make run/timer` at least on nova.

m-stein, 2017-09-14:
I'm at it...

m-stein, 2017-09-19:
@chelmuth Added fixes for the recent timer problems:
eb8e1bb timer test: absence of starvation under stress
dc48958 timer connection: fix mixing of time sources
b4658a7 signal: fix starvation by fast signal contexts

chelmuth, 2017-09-19:
Merged 8cd91dd6c for now only.

m-stein, 2017-09-25:
@chelmuth These commits should solve the issue:

cd04446 nic_router: dhcp server functionality
57fb27e nic_router: do link garbage collect on each packet
396df64 nic_router: handle ARP requests for foreign IPs
dc5a0c3 nic_router: use same packet log-format as nic_dump
8149fc8 nic_router: rework round-trip-time handling
750fcf9 net: make packet print-functionality configurable
aeff65b dhcp: extend options utilities
1d94092 net: clean up header fields and accessors
ec965db ethernet: rework type for ethernet type value
72e2fec nic_dump: use Duration instead of unsigned
ad3f39d nic_dump: measure time only if needed
bfb43ac bit_allocator: add dynamic version
423f1be nic_router: fix for unknown transport protocols

m-stein, 2017-09-26:
@chelmuth Limit for timer handling rate and a test whether it works: 
f2ed191 timer: limit rate of handling timeouts

chelmuth, 2017-09-27:
@m-stein I merged two commits to my private staging branch 53e243ceaf734a9f0dff04d61ba11c56ce73d1d8. Unfortunately, the timer test hangs on NOVA under Qemu. Could your please investigate?

m-stein, 2017-09-27:
@nfeske I've reworked the commits according to your suggestions. I moved the packet log to a new "nic_dump" lib so that it can be used in the nic_router as well and the include dir os/include/nic_dump marks the headers component local (like with init/child_policy.h). We could even name it net-dump to stay compliant with the naming of the net-stat lib. What do you think? I decided to remove the "common" print functionality completely from the packet classes in the net lib as I find it hard to tell what a user expects when handing over a packet to a log function.

8754a23 nic_router: dhcp server functionality
c3e243d nic_router: do link garbage collect on each packet
34059d6 nic_router: handle ARP requests for foreign IPs
1bf83cf nic_router: use same packet log-format as nic_dump
052eebc nic_router: rework round-trip-time handling
5d1fa4b net: enhance packet print, move to nic_dump lib
7977ff3 ipv4 address: conversion from and to integer
fa2c01f dhcp: extend options utilities
20023c2 net: clean up header fields and accessors
9e021b2 ethernet: rework type for ethernet type value
5a4a863 nic_dump: use Duration instead of unsigned
42c2dbe nic_dump: measure time only if needed
d64f9d4 bit_allocator: prepare for dynamic version
423f1be nic_router: fix for unknown transport protocols

m-stein, 2017-09-27:
@chelmuth I expect commit 53e243c "timer test: absence of starvation under stress " to work only with 75042dd554f7655d04b4d364fbd67cb6d82aa587 "signal: fix starvation by fast signal contexts"

m-stein, 2017-09-27:
I.e. please do not merge it before we did solve 75042dd "signal: fix starvation by fast signal contexts" correctly.

m-stein, 2017-09-27:
I'll rebase f2ed191 timer: limit rate of handling timeouts to staging...

m-stein, 2017-09-27:
@chelmuth @nfeske I reworked the commit for fixing signal starvation. The new version even simplifies the signal processing code a little. The commits should be applied in this order:

b59bf40 timer: limit rate of handling timeouts
2853339 timer test: absence of starvation under stress
682bb9a signal: fix starvation by fast signal contexts

nfeske, 2017-09-28:
@m-stein I am actually unhappy about your new revision.

>  I've reworked the commits according to your suggestions.

That is quite a stretch. In my comment on bfb43acc5fc3bc0d72a468e99192d35a7f28223f, I kindly asked you to keep base/include unchanged. You changed it anyway. I wonder, did you read the explanation of my rationale? Your modification increases code complexity while leaking implementation details that were formerly nicely hidden in the private class section. It does not bring any benefit for the base framework.

> I decided to remove the "common" print functionality completely from the ...

This frustrates me. You did not respond to my comment on 750fcf98997e6fa62bb67ee483d4cc22cb521059 but take single-handed decisions while downright ignoring large parts of my comment. I asked you to keep the protocol headers unchanged. You changed it anyway - for the worse. Now, with the new version, one can no longer log network packets while debugging nic_bridge or other components that may use the net headers. I suggested to untangle the formating of output and the log-policy logic. You instead promoted your entangled logging facility to become a public library - thereby inviting others to use it.

As I painfully experienced over the past year of cleaning up our code base, the approach of carelessly exposing component-specific implementation parts as public headers is wrong! We (me included) committed this sin repeatedly in the past. The proliferation of implementation parts (like os/include/init) throughout other components ultimately becomes a huge maintenance burden. It is easy to expose things. But it is extremely hard to reverse. Nowadays, I became extremely conservative with spilling half-baked implementation parts into publicly visible headers. E.g., even though I designed headers like os/src/init/buffered_xml.h or gems/src/app/menu_view/list_model_from_xml.h to be generally usable, I keep them local to the components that use them because I am not 100% sure that they will stand the test of time. I look at your logging facility from this perspective. Making it a publicly visible library is clearly a mistake as it introduces fairly ad-hoc decision of yours (like new terminology) and unproven patterns (like the particular way how configuration is expressed). As long as this facility is local to nic_dump, I don't care too much. But you are prematurely promoting it as a framework pattern, which I don't agree with.

I find this discussion exhausting. You did such a wonderful job with the NIC router. I do not understand why we enter bike-shedding territory over mundane issues like this debugging feature.

m-stein, 2017-09-28:
Removed all changes to public files from bfb43ac and 750fcf9:
db2c9d4 nic_router: dhcp server functionality
dc35171 nic_router: do link garbage collect on each packet
9ea2d81 nic_router: handle ARP requests for foreign IPs
708de0b nic_router: use same packet log-format as nic_dump
27d3976 nic_router: rework round-trip-time handling
66a3b6d nic_dump: configurable packet print
0c7ab5e ipv4 address: conversion from and to integer
394fde9 dhcp: extend options utilities
5eaba18 net: clean up header fields and accessors
dc09deb ethernet: rework type for ethernet type value
b36def9 nic_dump: use Duration instead of unsigned
5aa87e8 nic_dump: measure time only if needed
3da5d20 nic_router: fix for unknown transport protocols

nfeske, 2017-09-29:
@m-stein Thank you!

m-stein, 2017-09-29:
Three additions also thanks to the feedback by @MartijnVerschoor and @BorisMulder-CSL:
3b8ae84 nic_router: bad-protocol message only if verbose
b5b3bc2 fixup "nic_router: dhcp server functionality"
1fed495 fixup "nic_dump: configurable packet print"

chelmuth, 2017-09-29:
@m-stein From the patch series regarding timout rate limiting and starvation I merged 890e7c4b56a77bb6a6d5ec187d3711c703bf19d1 which is ea0a561 from you merge_to_staging branch. I removed the changes to the timer test as these still do not work (maybe due to the removal of the starvation patch.)

Please open a new issue for the starvation issue and attach 0b31a9e and 481155c to it for later merge.

chelmuth, 2017-09-29:
Some test with staging (890e7c4b56a77bb6a6d5ec187d3711c703bf19d1) reveal that run/timout hangs at *TEST 2: schedule multiple timeouts simultaneously* on all platforms I tested (Qemu nova, sel4, and okl4 as well as linux).

chelmuth, 2017-09-29:
I also tested your original patch series with https://github.com/chelmuth/genode/commits/dont_starve which also hangs.

m-stein, 2017-10-02:
I'm at it.

chelmuth, 2017-10-04:
I merged the commits and did not see any hangs so far, thanks! Are there any plans to completely merge the alarm lib with timeout?

Also please note, 3e0aa91ee1040293ed3b46da0a956af749ceaa11 on your branch changes test/timer only. Maybe you should merge it with bbcdbd6b7b094b7c9f284f96b9c062fa008c59a1.

m-stein, 2017-10-04:
Originally, https://github.com/genodelabs/genode/commit/3e0aa91ee1040293ed3b46da0a956af749ceaa11 is part of  890e7c4. The former is the test for the fix in the latter so I think it would be best to merge them again.

m-stein, 2017-10-04:
But anyway:
45ab34b timer test: rate limiting and no starvation

m-stein, 2017-10-04:
Regarding the alarm lib: Yes definitely but I don't know if it is already fine to do it.

chelmuth, 2017-10-04:
AFAIK there are three components that need to be cleaned up then: noux, netperf's special `alarm()` implementation, and timed_semaphore, with the latter being already deprecated.

m-stein, 2017-10-04:
I would like to proceed with the DHCP client for the NIC router before going into this clean-up topic. Is it critical?

chelmuth, 2017-10-04:
Not in the least, just shared my considerations.

chelmuth, 2017-10-05:
@m-stein please have a look at current staging (1950f8cba1376ef8c6781df975dc1d5d46f34c1b) and check if I missed a commit

m-stein, 2017-10-05:
Looks good to me. Thanks!

m-stein, 2017-10-06:
@chelmuth I've updated my merge_to_staging branch. It contains now your suggestions regarding "enum class" and a fixup 52ef610 for the recent problems with building base-hw. I suggest fb08b98 for the new staging HEAD. These are the new commits (based on b6fe9ff from staging):

fb08b98 depot: update recipe hashes
ac874b5 vfs: return distinguishable IDs when opening the root directory multiple times
e6c1484 nic_router: bad-protocol message only if verbose
c136668 fixup "nic_router: dhcp server functionality"
bdffbad fixup "nic_router: dhcp server functionality"
9c1ad6e nic_router: dhcp server functionality
3bbd4ac nic_router: do link garbage collect on each packet
7a4563c fixup "nic_router: handle ARP requests for foreign IPs"
0d14c21 nic_router: handle ARP requests for foreign IPs
61a0ca4 nic_router: use same packet log-format as nic_dump
8816909 nic_router: rework round-trip-time handling
1e01454 fixup "nic_dump: configurable packet print"
35fd59a fixup "nic_dump: configurable packet print"
4078336 fixup "nic_dump: configurable packet print"
bd9302a fixup "nic_dump: configurable packet print"
0ca407a nic_dump: configurable packet print
9c4d0fa fixup "dhcp: extend options utilities"
d91048a dhcp: extend options utilities
12988cb ipv4 address: conversion from and to integer
62dadb9 fixup "net: clean up header fields and accessors"
97f047b net: clean up header fields and accessors
ea5aaf8 fixup "ethernet: rework type for ethernet type value"
a587b28 ethernet: rework type for ethernet type value
78d8028 nic_dump: use Duration instead of unsigned
f81bd7d nic_dump: measure time only if needed
de84137 nic_router: fix for unknown transport protocols
e1d8497 timer test: rate limiting and no starvation
52ef610 fixup "signal: fix starvation by fast signal contexts"
2b58923 signal: fix starvation by fast signal contexts

m-stein, 2017-10-09:
@alex-ab 
a9cbb91 fixup "net: clean up header fields and accessors"

alex-ab, 2017-10-09:
@m-stein: thx,  added to staging

chelmuth, 2017-11-06:
May we close this issue?

m-stein, 2017-11-06:
Yep.

closed 2017-11-06 by m-stein