genodelabs/genode #2499: Port of PCG algorimithic RNG library

ehmry, 2017-08-19:
This patch ports a library for quickly producing statistically random numbers. Its output for a given initial state is reproducible so it must be seeded with an entropy source like Jitterentropy. I think it would be a useful noise source for tests (#2477) and is appropriate for the libports repository. I will also be using it in the world repository.

There is a talk on the internals that I remember being quite good.
http://www.pcg-random.org/posts/stanford-colloquium-talk.html

chelmuth, 2017-08-23:
The library builds fine on x86_64 with `make LIB=libpcg_random` but fails on arm and x86_32 with
```
In file included from .../contrib/pcg-c-ef987b4a48bed61732e791998c856d8326c5a897/src/lib/pcg-c/src/pcg-global-32.c:33:0:
.../contrib/pcg-c-ef987b4a48bed61732e791998c856d8326c5a897/include/pcg-c/pcg_variants.h:2178:33: error: ‘PCG_STATE_SETSEQ_64_INITIALIZER’ undeclared here (not in a function)
 #define PCG32_INITIALIZER       PCG_STATE_SETSEQ_64_INITIALIZER
                                 ^
.../contrib/pcg-c-ef987b4a48bed61732e791998c856d8326c5a897/src/lib/pcg-c/src/pcg-global-32.c:35:38: note: in expansion of macro ‘PCG32_INITIALIZER’
 static pcg32_random_t pcg32_global = PCG32_INITIALIZER;
                                      ^~~~~~~~~~~~~~~~~
.../repos/base/mk/generic.mk:52: recipe for target 'pcg-global-32.o' failed
```

ehmry, 2017-08-23:
Patched at b7015a2c7b9459df91d3eea06fde48b4124ce599 with a scenario to verify PCG32 output to be sure.

Ref https://github.com/imneme/pcg-c/pull/20

ehmry, 2017-08-23:
This has been corrected upstream so I am rebasing my branch with a fresh commit.

commit c7cb1309 (2017-08-24): port of PCG random number generator library

http://www.pcg-random.org/
http://www.pcg-random.org/using-pcg-c.html

Ref #2477
Fix #2499
closed 2017-08-30 by chelmuth