genodelabs/genode #2510: run tool support for mutible FAT boot images

ehmry, 2017-09-03:
I have a appliance-style run scenario that writes back to the boot media so I would like some run modules that would allow building and testing images for USB sticks with QEMU.

My initial implementation creates a bare yet bootable FAT image with Syslinux without a using partition table. My plan is to next create an 'image/mbr' module that would wrap ``[run_dir].partition`` with a MBR partition containing the Syslinux boot sector.

ehmry, 2017-09-03:
Implemented at 37fb0af7b5cedcfeacc252f52ead73656e0baab1.

chelmuth, 2017-09-27:
@ehmry and @alex-ab could you both please discuss how to merge this feature with #2526 which is almost ready for staging?

ehmry, 2017-09-27:
I still want to be able to build images for USB sticks but I am unsure that my patch here is best way to do this, so I am fine with yielding to @alex-ab's patch.

alex-ab, 2017-09-27:
In principle the both commits could coexist.

nfeske, 2017-12-13:
I wonder, what is the state of this issue? Basic support for FAT on boot media was added to the run tool in the form of the image/uefi backend (based on mtools). It may not address @ehmry's use case though. What would be the appropriate steps for this issue?

ehmry, 2017-12-13:
I will close this for now. What I was interested in was making mutable images for USB sticks and then booting QEMU from them.

closed 2017-12-13 by ehmry