genodelabs/genode #2511: init does not halt exiting children

ehmry, 2017-09-05:


ehmry, 2017-09-05:
Fixed at 7accb28f12aa7171bc67b31842d737cb69f64310.

BorisMulder-CSL, 2017-09-11:
For my own work, I implemented something similar into init, but also made a feature that causes the state report generated by init to contain a state="exited" attribute for the child if that child has exited. 
So whenever a child goes to STATE_EXITED, init will generate a new report with this state attribute.
This way, other components with access to this report can see that a certain child has quit.
Perhaps this would also be a nice feature of init? Currently the report contains a state="incomplete" attribute when the child has gone inactive, doing the same for exited children would make sense here.

ehmry, 2017-09-11:
@BorisMulder-CSL That is a good idea, I hadn't thought of that.

nfeske, 2017-09-18:
@BorisMulder-CSL I agree with the direction you described and would very much appreciate a corresponding patch, preferably accompanied with a test sequence for init.run. ;-)

nfeske, 2017-12-22:
The `exited` attribute is now supported by init, thanks to commit eae0445cd8af2482ea7f94b8f25853b7670938b4. May we close this issue?

ehmry, 2017-12-22:
Fine by me.

closed 2017-12-22 by ehmry