genodelabs/genode #2512: race condition in socket_fs socket binding

ehmry, 2017-09-05:
Some socket_fs control files are not present until a another file is written to, this is the case with ``bind`` and ``listen``. The VFS does not currently sync writes which can allow an application to open the ``listen`` file before a write to the ``bind`` file completes, which must occur before ``listen`` is available.

As a side note, the VFS server can return uninitialized file statuses for missing files, which can make missing files appear to exist with arbitrary types.

ehmry, 2017-09-06:
Working with the lwIP plugin made me realize my solution was wrong, the better fix is to leave the current write and sync behavior in place and simply sync the bind control file before listen.

Fixed at e072e62846a948813c5d772864dbb59a35034c63 and 7bebd6725ce329d36b850e86ba88a914efac0b52.

I had considered using ``sync`` to block lwIP writes until TCP acknowledgments had cleared, but delayed ACKs would halt the stream and would probably be terrible for performance in typical cases.

chelmuth, 2017-09-06:
Thanks for the investigation. I merged 8a726e676a6fb07f9a3035e80b32ec233b2f118c and 4d601560b42d203a96d74d8f23ece624f2c1b1b5 to staging.

commit 8a726e67 (2017-09-06): libc: sync bind sock control file before continuing with listen

Fix #2512
commit 577c9b1a (2017-09-07): libc: sync bind sock control file before continuing with listen

Fix #2512
commit 8207fb8d (2017-10-05): libc: sync bind sock control file before continuing with listen

Fix #2512
closed 2017-10-19 by chelmuth