genodelabs/genode #2513: tool: extend xmllint by Genode XML schema to check (offline) for known Genode config XML nodes/attributes

alex-ab, 2017-09-06:


chelmuth, 2017-09-06:
This is nice! Does the solution work with inline content like in *netperf_lwip_wifi.run*?

alex-ab, 2017-09-06:
I did not specify what the config node elements/attributes of the start nodes, so yes it will work (by ignoring/not checking). Of course, here we would need to add some component specific .xsd extensions, if you would like also to check that.

alex-ab, 2017-09-06:
I restricted the checking for now just to the install_config proc, otherwise other .config file would also be checked, which contains e.g. for the launcher launcher specific config formats, not understood for the moment.

chelmuth, 2017-09-06:
I did not want to extent the scope beyond the init configuration, but I want to make sure that inline content does not
- break the check with a failure
- disable the check completely due to weird syntax
Thanks anyway for the feature and the explicit limitation to the install step.

chelmuth, 2017-09-08:
Are your patches fit for merging already?

alex-ab, 2017-09-08:
In principle yes.

chelmuth, 2017-09-08:
Okay, as soon as you mark it `fixed` I'll merge the commits (preferably merged into one).

commit 121e5879 (2017-09-22): tool: add genode xml schema to xmllint check

Fixes #2513
commit 4eeb09f2 (2017-09-25): fixup "tool: add genode xml schema to xmllint check"

Fixes #2513
commit b3e0cd5e (2017-09-25): fixup "tool: add genode xml schema to xmllint check"

Fixes #2513
closed 2017-10-19 by chelmuth