genodelabs/genode #2519: File_system: directory entries not showing

BorisMulder-CSL, 2017-09-12:
I have a problem with rump_fs on Genode 17.08: When I have a rump_fs server running and a client reads from a directory (to see which files are inside), it randomly does not show some files.
This happens for instance when I run noux and do a "ls" on the directory, or when I just log all the directory entries that I could find in a custom client.

When I just run ls (or some other program in noux such as cat) on one of the missing files specifically, it does list the file. Also, opening one of the missing files through the file system session interface just works.

It does seem that given a certain content of the directory, always the same files will be invisible, but when the content is changed, different files might or might not become invisible. There is always a different number of files invisible to ls, and there does not seem to be a cap on the number or sizes of files that are listed correctly.

Also perhaps worth noting is that in my scenario, I have multiple clients that have read-only access to the file system, and two clients that have r/w access, though they do not write the same files simultaneously.

Things I have tried: running fsck, rebooting, booting linux, changing something in the directory, reverting to 17.05, but in vain. Fsck does not detect any problems. On linux or on 17.05, I always see all the files correctly; the problem only happens under Genode 17.08.

I have not been able to reproduce this issue consistently on different computers or create a meaningful test scenario for debugging purposes. Does anyone have an idea what could be wrong?

chelmuth, 2017-09-12:
This is a new issue to me and I suspect it could be a regression with asynchronous read/getdirentries. Could you please try to investigate if https://github.com/genodelabs/genode/blob/17.08/repos/libports/src/lib/libc/vfs_plugin.cc#L605 is triggered by some corner case and returns a zero-read to early?

BorisMulder-CSL, 2017-09-12:
When I put a print statement between line 606 and 607 there, it is not triggered as far as I can see.
However, now somehow ls randomly does show all files.
The following code does not list all files though:

`for ( unsigned i = 0; ; i++ )
{
	using Entry = File_system::Directory_entry;
	Entry entry;
	size_t size = File_system::read( _fs, dir, ( char * ) &entry, sizeof( Entry ), i*sizeof( Entry ) );

	if ( size < sizeof( Entry ) ) {
		break;
	}
	Genode::log(entry.name);
}`

chelmuth, 2017-09-12:
IMO your example discharges the VFS code. Do you have a ready run scenario with a native Genode component using your code you could provide for further investigation? It could also be helpful to have some hints how the ext2 image should be generated (number of files, directory structure).

nfeske, 2017-12-13:
We could reproduce the problem in the sculpt scenario. It is fixed with commit 9c852c750a034a073218633554d020f3ade6ea04.

closed 2017-12-13 by nfeske